bambulab/BambuStudio #3711: Wrong gap infill after slicing

gregory353a, 2024-03-12:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 

### Additional system information

n/a

### Printer

Bambu Lab X1 series

### How to reproduce

1. Slice project.
2. Go to preview

### Actual results

On sliced model there are 3 horizontal gap infills on right holes heigh. That infills causing wrong extrusion.
[image]
[image]

### Expected results

Sliced model without wrong gap infills 

### Project file & Debug log uploads

[skadis_box.zip](https://github.com/bambulab/BambuStudio/files/14578933/skadis_box.zip)
[BambuStudio.zip](https://github.com/bambulab/BambuStudio/files/14578949/BambuStudio.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SaltWei, 2024-03-18:
可以分析一下是不是维诺图的问题 @XunZhangBambu

SaltWei, 2024-08-06:
fixed in 1.9 version

closed 2024-08-06 by SaltWei