bambulab/BambuStudio #3712: Multiple instances of BambuStudio hang when one instance hangs

ansonl, 2024-03-13:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

Win10

### Printer

Bambu X1C

### How to reproduce

1. Open two instances of BambuStudio
2. Import a large 3MF file into first instance
3. Try to switch to and interact with the second instance of BambuStudio

### Actual results

Second instance of BambuStudio will be frozen until first instance unfreezes from importing 3MF file even though it is not processing anything related to the first instance. 

### Expected results

Second instance of BambuStudio should not lag or hang. It should be independent of the first instance thread. 

### Project file & Debug log uploads

example 3MF file that takes a while to load (I usually work with 3MF up to 200-400mb in size)
[MD-linear-dual-v2-bs.zip](https://github.com/bambulab/BambuStudio/files/14582067/MD-linear-dual-v2-bs.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ansonl, 2025-02-20:
This hanging behavior with multiple Bambu Studio instances is still present in the latest version

lanewei120, 2025-02-20:
Hello @ansonl

could you help to give a video to help us understand this process?
and please open the task manager at the same time

thanks

ansonl, 2025-02-22:
@lanewei120 

I have attached a recording to show the issue. The left window is unresponsive when the right window is busy.

[Animation.zip](https://github.com/user-attachments/files/18919737/Animation.zip)