bambulab/BambuStudio #3714: Dialog Location

gothickitty93, 2024-03-13:
> [image]

Where do I find this UI? I have no idea where this dialog box is in either Bambu Studio or OrcaSlicer, and it would really help me out as I'm trying to connect to my printer via my personal VPN

_Originally posted by @gothickitty93 in https://github.com/bambulab/BambuStudio/issues/1285#issuecomment-1995377027_

lanewei120, 2024-03-18:
it is on printer's screen gui

gothickitty93, 2024-03-18:
So this dialog box cannot be accessed manually it seems. It only shows up if there is a failed connection attempt to the printer. I feel like we should be able to access this dialog whenever we need to manually.

lanewei120, 2024-03-19:
it should always be there
however, as the gui layout has been changed
I attached the current picture here
[image]

gothickitty93, 2024-03-19:
I think you misunderstood me. I’m talking about this dialog box being accessible in Bambu Studio, allowing the user to type in the IP and access code of their printer manually.

lanewei120, 2024-03-19:
oops, sorry for the misunderstanding

the dialog on bambu studio only popup when there is a failed connection attempt currently

timothyzalusky, 2024-05-14:
This is a feature request that is also tied to adding the printer to Bambu Studio Via IP address and Bambu Lab really is dropping the ball here.  We need this feature and I don't know why they have not fixed this yet.