bambulab/BambuStudio #3726: Cannot pause print when print sequence by object

tcurtin, 2024-03-14:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Sonoma 14.1.2

### Printer

Bambu Lab P1S

### How to reproduce

1. Switch to Plate Settings/Print sequence to By object
2. Unable to right-click on layer slider and add a pause

### Actual results

Unable to set a pause when printing by object

### Expected results

Should be able to set a pause when printing by object

### Project file & Debug log uploads

Happens with any project I've tried.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoCuul, 2024-04-07:
I have the same issue, it's infuriating that this limitation exists.

SoCuul, 2025-08-31:
This still has not been addressed @lanewei120

fcrozat, 2025-12-05:
Still valid