bambulab/BambuStudio #3727: Unable to open calibration dropdown on second material

ThePieMonster, 2024-03-15:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Unable to open calibration dropdown on second material. 

As seen in the first image, there are two calibrations there were made.

The second material dropdown will not open to allow me to select this.

[image]
[image]
[image]

### Actual results

No results

### Expected results

Dropdown works

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-03-27:
Because the PA k value is related to the material, the k value calibrated from PLA Basic material can only be used for PLA Basic material and cannot be used for the PLA Metal material. Are your PA profiles both calibrated from PLA Basic material?

ThePieMonster, 2024-03-27:
> Because the PA k value is related to the material, the k value calibrated from PLA Basic material can only be used for PLA Basic material and cannot be used for the PLA Metal material. _Are your PA profiles both calibrated from PLA Basic material?_

Check the images again.

There is a k profile for basic and a k profile for metal. 1 for each.

zhimin-zeng-bambulab, 2024-03-27:
[image]

Could you confirm that the filament presets you selected when sending calibration are correct? If you are sure it is correct, could you upload your log file at that time.

ThePieMonster, 2024-04-02:
I have since changed out filaments types quite a bit. Not able to reproduce this at the moment.

zhimin-zeng-bambulab, 2024-04-02:
Close as unable to reproduce.
If you encounter this problem again later, you could upload the the log file and record the preset information selected at that time. Thanks!

closed 2024-04-02 by zhimin-zeng-bambulab