bambulab/BambuStudio #3728: Unviewable model in Painting Mode if the model is moved below the bed

Kingsidorak, 2024-03-16:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1S

### How to reproduce

Move a model below the build plate view

### Actual results

parts under the build plate make it nearly impossible to see details in the painting view

### Expected results

The program should detect parts under the build plate, and create a line where the model cannot be painted, or cut the model

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-03-18:
In paint mode, the complete model should be displayed by default. Do I understand correctly?

Haidiye00, 2024-03-26:
Please use the translation tool to drag the object above the ground

Kingsidorak, 2024-03-26:
That suggestion is not fixing this problem that happens within painting. It is making me undo changes to the model so I can paint it, and then I would have to move it back down again, all because the feature is bugged; which I am already doing because the feature is bugged

Haidiye00, 2024-03-27:
Let me see if our company has a computer with this resolution. If we can find it, we can make some changes

Haidiye00, 2024-05-15:
Design like this. Sorry, we won't make any changes at the moment. You can move the object onto the hot bed.

closed 2024-05-15 by Haidiye00