bambulab/BambuStudio #3729: Custom Filaments Not Visible in AMS Filament Selection

kiltless-wonder, 2024-03-16:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10, Windows 11, Android 

### Printer

Bambu A1

### How to reproduce

Custom Filament [Manufacturer] [PLA] [Series] created in filament menu
Custom filament set on Bambu Handy
Sync Materials in prepare tab of Studio to show correct filament and manufacturer
Observe AMD material under Device Tab now blank, custom material not available to select
[image]
[image]

https://github.com/bambulab/BambuStudio/assets/59786869/2b5ebe64-547e-4dbc-92bb-0054526d57ef

### Actual results

Device AMS material selection does not show custom filament on bambu studio
Device AMS material section does show custom filament on bambu handy

### Expected results

Custom filament profiles should be available in AMS material selection, filament options should sync between Bambu Handy and Studio

### Project file & Debug log uploads

[debug_Fri_Mar_15_20_00_34_6016.log.zip](https://github.com/bambulab/BambuStudio/files/14621934/debug_Fri_Mar_15_20_00_34_6016.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

kiltless-wonder, 2024-03-19:
Update: On the PC that the custom filament was created, after restarting the PC and waiting 12+ hours the filament is correctly assigned to AMS Slot;
[image]

All other computers with Bambu Studio have been restarted and now show an error when re-syncing the AMS 
[image]

and display the AMS slot as blank 
[image]

kiltless-wonder, 2024-03-19:
Update 2: Printer does not show Custom Filament. Arbitrarily sets to another material, and color when Custom Filament is assigned to AMS slot via Bambu Handy. Current Materials set in Bambu Handy (note the question mark in the filament color when you first click into the material)
https://github.com/bambulab/BambuStudio/assets/59786869/1c704a7b-fc2c-4821-bb05-b49ba805d9b5

Corresponding Printer Materials Page; 
https://github.com/bambulab/BambuStudio/assets/59786869/d469cb67-9114-4545-a37d-3e94e8b68621

If you watch the behavior frame by frame as the "Edit Menu" opens you will see correct material and incorrect color transition to incorrect material and incorrect color. 

[image]
[image]
[image]
[image]

DanBao-Bambu, 2024-03-24:
I'm sorry for the inconvenience caused to you. As far as I know, there is currently no printer version available for A-series machines that supports custom Filament.

kiltless-wonder, 2024-03-25:
@DanBao-Bambu Any idea when that is a planned feature? 

In the meantime I suppose I can control temperature via the filament settings in studio, but I must be able to set the custom K-value. I can see the ability to set on the printer but do not see any documentation to do so
[image]