bambulab/BambuStudio #3730: No 3mf Preview in Explorer

gvogele, 2024-03-16:
### Bambu Studio Version

Latest

### Where is the application from?

Bambu Lab Official website

### OS version

W11

### Printer

BambuLab X1C

### How to reproduce

open Explorer

### Actual results

no Preview

### Expected results

working Preview

### Project file & Debug log uploads

No File

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-03-18:
@gvogele is the other 3mf not from bambustudio working correctly?

we have written the thumbnail into _rels/.rels
which should be supported by windows

gvogele, 2024-03-18:
Thank you, i have found now that some Program changed the REG.
I added the entries and now everything is working.
I added the two REG-Files to correct the Error. I am working on W11, not shure if it works on W10. Pleas rename to *.reg before use.
[3mf.txt](https://github.com/bambulab/BambuStudio/files/14643181/3mf.txt)
[Preview.txt](https://github.com/bambulab/BambuStudio/files/14643182/Preview.txt)

After Installing the Two Files you have to choose one 3mf File an right Mousebutton > Settings and change to Bambu Studio.
Reboot maybe required.
Greetings

lanewei120, 2024-03-19:
Glad to hear that!

thanks for the sharing~

closed 2024-08-08 by lanewei120

scubi, 2025-01-16:
Can you please help, unable to view my 3MF files in Explorer or Bambu lab studio, thanks