bambulab/BambuStudio #3731: 无法发送切片文件给打印机，显示网络问题，但是直接选择打印单盘就可以正常发送打印

name20010723, 2024-03-16:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

Windows11

### Printer

P1S

### How to reproduce

切片后选择发送给打印机

### Actual results

无法发送切片文件给打印机，显示网络问题，但是直接选择打印单盘就可以正常发送打印

### Expected results

可以正常发送给打印机

### Project file & Debug log uploads

[jet.zip](https://github.com/bambulab/BambuStudio/files/14624266/jet.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

walterwongbbl, 2024-03-18:
可以提供一下Studio的log日志吗

name20010723, 2024-03-18:
> 可以提供一下Studio的log日志吗

[log.zip](https://github.com/bambulab/BambuStudio/files/14633321/log.zip)