bambulab/BambuStudio #3732: Text edit crash

Gateszs, 2024-03-16:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 23h2

### Printer

Bambu Labs P1S

### How to reproduce

assembly plate 2
category  edit text
crash
 

### Actual results

[image]
[image]

### Expected results

Edit text

### Project file & Debug log uploads

[Big+Oscar+-+With+Base+Plaque (1).zip](https://github.com/bambulab/BambuStudio/files/14625576/Big%2BOscar%2B-%2BWith%2BBase%2BPlaque.1.zip)
[bambu-studio.exe.6692.dmp](https://github.com/bambulab/BambuStudio/files/14625578/bambu-studio.exe.6692.dmp)
[crash_Sat_Mar_16_01_19_42_0_done.log](https://github.com/bambulab/BambuStudio/files/14625579/crash_Sat_Mar_16_01_19_42_0_done.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-03-18:
I have identified the cause of the problem and solved it. Thank you for discovering the issue

Gateszs, 2024-03-26:
Is there an updated file?

Haidiye00, 2024-03-27:
V1.9 Beta version has been released, you can download it on GitHub

closed 2024-05-06 by Haidiye00