bambulab/BambuStudio #3733: Bambu studio crashing after opening

Gaijindragon, 2024-03-17:
### Bambu Studio Version

1.8.4.5.1

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

cpu:12th gen intel core i5-1235u
ram: 32gb

### Printer

bambu labs x1 carbon

### How to reproduce

Open about 12-13 .3mf files then happens anytime application opens

### Actual results

after opening files bambu studios will display the error "unhandled unknown exception: terminating application

### Expected results

application should open

### Project file & Debug log uploads

[crash_Sat_Mar_16_20_58_45_0.zip](https://github.com/bambulab/BambuStudio/files/14626145/crash_Sat_Mar_16_20_58_45_0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Haidiye00, 2024-03-18:
Opening one 3mf has no problem, but opening more than 10 3mfs causes a crash? Is that so?

zloyded, 2024-03-24:
Also crash on MACos while opening after confirm or decline notification about fill gyroid density not support with tree supports

Dedoque, 2024-03-24:
> Also crash on MACos while opening after confirm or decline notification about fill gyroid density not support with tree supports

Same stufff.
remove and install new one. Remove from `Application/` and `Application Support` dirs

Haidiye00, 2024-05-06:
Hello, has your issue been resolved?

Dedoque, 2024-08-26:
No. I've switched to another slicer and now alright!

closed 2024-08-27 by Haidiye00