bambulab/BambuStudio #3735: Please add a Feature for Re-Arrange plates

SvenBudak, 2024-03-18:
The title says already everything... there should be really a feature that allows rearranging the order of print plates.

dcArock, 2025-01-15:
YES! Especially because 3MF Uploads to the MakerWorld.com platform retains the PLATE ordering with its matching names.

Here in my situation I want to offer users a simple profile where they can print multiple sizes of an item:
1x2 / 1x3 / 1x4 up to 4x4
But someone requested I add a 3x5
So now to update my project file, I currently need to delete all plate after the size I want to add, because I can't simply re-arrange plates while keeping their names.

Quality of life improvement and I can't imagine it's that hard of a feature to implement.

[image]

pmoore, 2025-01-22:
+1 to this feature request.  Would be a great QoL improvement and time saver!

Jamyn, 2025-08-15:
Where is this feature on the Bambu priority list? (Is there an internal case number tracking this request?) There is a multi-year, long [discussion thread](https://forum.bambulab.com/t/feature-request-reorder-plates-in-a-project/46426/88) asking for this in the Bambu forums. Could this be increased in priority?