bambulab/BambuStudio #3737: 无法在设备AMS和外挂设置中找到自定义耗材

ketoohome, 2024-03-19:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Window 11

### Printer

Bambu A1

### How to reproduce

1. 创建自定义耗材，三绿TPU 透明橙，打印温度范围195~205°
2. 在设备页面，选择选项更改AMS或外挂材料。

### Actual results

无法在设备AMS和外挂设置中找到自定义耗材

### Expected results

允许选择自定义参数的耗材

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14629661/log.zip)
[KK_Twins11(4).zip](https://github.com/bambulab/BambuStudio/files/14629664/KK_Twins11.4.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

DanBao-Bambu, 2024-03-24:
I'm sorry for the inconvenience caused to you. As far as I know, there is currently no printer version available for A-series machines that supports custom Filament.

wuyingnan, 2024-04-13:
> I'm sorry for the inconvenience caused to you. As far as I know, there is currently no printer version available for A-series machines that supports custom Filament.

Do you have any roadmaps for A1 mini custom filament?

DanBao-Bambu, 2024-04-19:
The operating logic should be the same, but currently A1 does not support custom Filament settings.