bambulab/BambuStudio #3738: Bambu Handy like skip object

gregory353a, 2024-03-19:
Simply I am asking for feature to skip object from Bambu Lab studio.

We are using Bambu printer in foundation, Very often foundation is empty, and only access to printer is through Bambu Lab Studio app behind remote connection. Sometimes during print one objects fails and it is pointless for printer to continue work on it. Due to limited access to login credentials we don't have workaround for our members to skip object.

thebadluck, 2024-10-13:
I agree this feature would be very helpful in Bambu Studio.

jlo1975, 2024-11-20:
we can do everything in bambu studio except this thing.....
So yes!
Not to have the option in bambu studio to mark failed object to be skipped is a huge design mistake.