bambulab/BambuStudio #3739: [Windows10] Camera video is not disaplaying. Status is "Loading" w/o "failed" or error code

maxmailbox, 2024-03-19:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

NVIDIA card only

### Printer

P1S

### How to reproduce

1. Decide tab
2. Start Camera

### Actual results

No video/image disaplying
"Loading..." label. No "failed", no error code
Firewall is OFF
Camera is working in phone app

https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view page is missing this case

### Expected results

Show video

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-03-21:
Try opening Windows Media Player directly. 
If you can't open it, uninstall it in the optional functions and reinstall it. 
If the installation fails, please refer to the wiki to install the offline version of WMP.
If there are any other issues, please provide Studio logs.

maxmailbox, 2024-03-21:
1. Windows Media Player is OK and playing files
2. As mentioned in bug description wiki does not cover this case
3. Nothing useful in logs on Play press:
...
2024-03-21 11:41:18.919083[Thread 0x0000071c]:MediaPlayCtrl::TogglePlay
2024-03-21 11:41:18.919083[Thread 0x0000071c]:MediaPlayCtrl::SetStatus: Stopped.
2024-03-21 11:41:18.926081[Thread 0x00003568]:MediaPlayCtrl: start stop
2024-03-21 11:41:18.927083[Thread 0x00003568]:MediaPlayCtrl: end stop
2024-03-21 11:41:18.927083[Thread 0x0000071c]:MediaPlayCtrl::onStateChanged: 0, last_state: 3
2024-03-21 11:41:18.927083[Thread 0x0000071c]:MediaPlayCtrl::onStateChanged: 0, last_state: 3
...

shaneo88, 2024-04-16:
I am also experiencing exactly this, and I am coming here after seeing that the wiki does not mention this situation. 

I can view the camera perfectly on my phone in Bambu Handy, but on my PC and my laptop the camera just says "Loading..." and never stops unless I change tab. It never throws an error.

cmguo, 2024-04-16:
@shaneo88 @maxmailbox 
Please help reproduce the problem and provide the log file of the reproduce time point.

bambu123, 2024-04-16:
@maxmailbox @shaneo88 
Please help reproduce the problem and provide the log file of the reproduce time point.

maxmailbox, 2024-04-17:
@bambu123 
Problem is 100% reproducible
Logs were already provided in the upper comment

bambu123, 2024-04-17:
> @bambu123 Problem is 100% reproducible Logs were already provided in the upper comment

Can you upgrade the network plugin? The log of this version of the plugin is incomplete. Reproduce it after the upgrade and provide new logs.
It's best to provide the entire log file. If you have any privacy concerns, you can send it to me via email.

maxmailbox, 2024-04-17:
> Can you upgrade the network plugin? The log of this version of the plugin is incomplete. Reproduce it after the upgrade and provide new logs. It's best to provide the entire log file. If you have [...]

Sure, I can upgrade plugin if u post the link

bambu123, 2024-04-18:
> Sure, I can upgrade plugin if u post the link

Did you not receive the prompt for the plugin upgrade? You can also delete the plugin and Studio will download it again by itself.
Plugin directory:% APPDATA%\ BambuStudio\ plugins

maxmailbox, 2024-04-18:
I did not received any prompt for plugin upgrade
After deleting BambuStudio\plugins, studio has downloaded exactly the same binaries: bambu_networking.dll, BambuSource.dll,
live555.dll
Logs:
2024-04-18 11:20:21.412879[Thread 0x00003c68]:MediaPlayCtrl::TogglePlay
2024-04-18 11:20:21.413866[Thread 0x00003c68]:MediaPlayCtrl::SetStatus: Initializing...
2024-04-18 11:20:21.434707[Thread 0x00003c68]:MediaPlayCtrl: bambu:///A7UD************111A?authkey=********&passwd=******&region=us&device=01P0*******0986&version=01.08.04.06&dev_ver=01.05.02.00
2024-04-18 11:20:21.434707[Thread 0x00003c68]:MediaPlayCtrl::SetStatus: Loading...
2024-04-18 11:20:21.435705[Thread 0x0000529c]:MediaPlayCtrl: start load
2024-04-18 11:20:21.435705[Thread 0x0000529c]:wxMediaCtrl2::Load: clsid {233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA} path C:\Users\user\AppData\Roaming\BambuStudio\plugins\BambuSource.dll
2024-04-18 11:20:21.440122[Thread 0x0000529c]:MediaPlayCtrl: end load
2024-04-18 11:20:21.466124[Thread 0x00003c68]:MediaPlayCtrl::onStateChanged: -1, last_state: 5

bambu123, 2024-04-18:
> I did not received any prompt for plugin upgrade After deleting BambuStudio\plugins, studio has downloaded exactly the same binaries: bambu_networking.dll, BambuSource.dll, live555.dll Logs: [...]

This log only shows the start of loading, not the process and result of loading.

maxmailbox, 2024-04-18:
If there would be result of loading - this bug would never appear:

2024-04-18 11:20:21.440122[Thread 0x0000529c]:MediaPlayCtrl: end load
2024-04-18 11:20:21.466124[Thread 0x00003c68]:MediaPlayCtrl::onStateChanged: -1, last_state: 5
2024-04-18 11:20:21.474935[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:22.469248[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:23.473073[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:24.468790[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:25.452573[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:26.484221[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:26.621747[Thread 0x00002530]:process_task{ type:Backup, id:0, path:, object:0, extra:2} and interval = 10
2024-04-18 11:20:26.635740[Thread 0x00003c68]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:2} and interval = 10
2024-04-18 11:20:27.454518[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:28.464374[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:29.463012[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:30.463676[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:31.465436[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:32.455017[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:33.459489[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:34.460447[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:35.461692[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page
2024-04-18 11:20:36.468723[Thread 0x00003c68]:Slic3r::GUI::PrintingTaskPanel::market_scoring_hide hide market scoring page

bambu123, 2024-04-18:
> If there would be result of loading - this bug would never appear:
> 
> 2024-04-18 11:20:21.440122[Thread 0x0000529c]:MediaPlayCtrl: end load 2024-04-18 11:20:21.466124[Thread 0x00003c68]:MediaPlayCtrl::onStateChanged: -1, last_state: 5 2024-04-18 11:20:21.474935[Thread 

It seems that the player is not working at all. You can try the WMP offline installation package in the wiki.

maxmailbox, 2024-04-18:
Its mentioned twice, in bug description and in comment that "Windows Media Player application is playing video OK"

bambu123, 2024-04-18:
> Its mentioned twice, in bug description and in comment that "Windows Media Player application is playing video OK"

The new version of WMP does not support the API of COM interface, so the old version is needed. We are excluding the dependency on WMP and expect version 1.10 to be completed.

maxmailbox, 2024-06-23:
Does the "Bambu Studio" works on Windows 11 with latest version of WMP ?

bambu123, 2024-06-24:
> Does the "Bambu Studio" works on Windows 11 with latest version of WMP ?

  No, the latest WMP is not a COM mechanism anymore.

fahaaaxiki, 2024-12-06:
Hi， we have solved this issue in our latest Public version：[1.10.1 Public Release](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50), you can update to this version to check if it works

maxmailbox, 2024-12-09:
Thanks, it works

closed 2024-12-09 by maxmailbox