bambulab/BambuStudio #3740: Failed to generate G-code for Custom Filament and Support Interface

kiltless-wonder, 2024-03-19:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10, Windows 11

### Additional system information

Multiple GPUs, CPUs, systems specs (5+ PCs) 

### Printer

Bambu A1

### How to reproduce

Create custom filament ( [Manufacturer] PLA [series] for three different materials all failed) 
Create filament settings for custom filament
Assign custom filament to correct AMS slot
Import STL/3MF (geometry) 
Process > Objects > Set model to custom filament settings
Process > Global> Select any process profile 
Enable Supports
Change Support interface to any material other than model material 

### Actual results

Error  prompt 

"Failed to generate G-code for invalid custom G-code.
change_filament_gcode Parsing error at line 25: Variable does not exist M620.10 A1 F[new_filament_e_feedrate] L[flush_length] H[nozzle_diameter] T[nozzle_temperature_range_high]                 
Please check the custom G-code or use the default custom G-code." 

L[flush_length] is variable of error

### Expected results

Custom filament should allow a different support interface.

### Project file & Debug log uploads

Video is cut in half to respect file size upload limit
https://github.com/bambulab/BambuStudio/assets/59786869/325ac8b9-9b1f-40f6-9989-2ba73d3d969e
https://github.com/bambulab/BambuStudio/assets/59786869/80a99282-320a-422b-8603-67cb9c3e6d9e
[Project File Error.zip](https://github.com/bambulab/BambuStudio/files/14653128/Project.File.Error.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/14653145/log.zip)

Possibly related to [Issue 3729](https://github.com/bambulab/BambuStudio/issues/3729)

### Checklist of files to include

- [X] Log file
- [X] Project file

albertcai101, 2024-04-12:
Fix for me: Click "Flushing Volumes" at the top left. For some reason new filaments reset the diagonal values to null for me.
[image]

tomLadder, 2024-04-28:
Same issue with AMS

XunZhangBambu, 2024-04-29:
@tomLadder @kiltless-wonder  This has been fixed in beta4 version

closed 2024-05-06 by XunZhangBambu

kiltless-wonder, 2024-05-06:
Thanks for addressing! I'll have to test this week when I can get the beta installed

XunZhangBambu, 2024-05-06:
> Thanks for addressing! I'll have to test this week when I can get the beta installed

The beta4 version has been released. You can get it through the link below:
https://github.com/bambulab/BambuStudio/releases/tag/v01.09.01.58