bambulab/BambuStudio #3741: Layer Times in Preview are showing wrong color and do not match legend color values!

odysoftware, 2024-03-19:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Go to Preview and Look at Layer Times visualization. You will see that the color's shown will not match the numbers for the layer times.

You can see an example here: https://github.com/bambulab/BambuStudio/issues/2046#issuecomment-1635892008

It shows 24s layer time, but the visualization color magenta indicates around 1-3s (as seen from the legend info)! It should be colored yellow to green to be in the 20s+ range.

### Actual results

Layer times visualization color matches wrong layer times values in legend.

### Expected results

Color in layer times matching the correct legend color with the correct value.

### Project file & Debug log uploads

not needed.

### Checklist of files to include

- [X] Log file
- [ ] Project file

odysoftware, 2024-03-20:
Added example 3MF file -> [test layer times.zip](https://github.com/bambulab/BambuStudio/files/14663676/test.layer.times.zip)
[image]

Haidiye00, 2024-03-26:
I have identified the cause and resolved it. Thank you for discovering the bug
[image]

odysoftware, 2024-03-26:
Thanks a lot 👍

closed 2024-05-06 by Haidiye00