bambulab/BambuStudio #3742: Error "unable to print filaments with big temperature difference"

kmeldner, 2024-03-19:
### Bambu Studio Version

1.8.4.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1 Carbon

### How to reproduce

1) import model to print with 2 filaments
2) set filaments PA-CF and PLA
3) assign filaments to model parts

Error appears

### Actual results

Error appears and prevents user from slicing.
[image]

### Expected results

No error.
I want to print what I want without having to go through complicated workarounds.

For the time being I have found one not so complicated workaround which is to set print sequence to "by object" in which case error does not come up.
The other one is to define same type of filament with different print settings suitable for totally different type and physically feed what you want making AMS think it is suing something totally different.

### Project file & Debug log uploads

[mocowanie silnika 8NmY.zip](https://github.com/bambulab/BambuStudio/files/14654953/mocowanie.silnika.8NmY.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

kiler129, 2024-03-20:
This is not a bug - the slicer is behaving exactly as it supposed to. You have multiple solids (that I'm guessing should be boolean-merged in your CAD):

[image]

If you assign completely different filaments to different parts of the same model, these filaments cannot be too far in terms of temperature settings. If one filament is set to e.g. 250°C and another is 200°C you're going to clog the nozzle eventually, which the error is precisely informing you about. This is because during change the nozzle will be too cold to flush the higher temp filament fully.

You can print two dissimilar materials in "by object" mode, and it's not a workaround but a feature. However, these cannot be a single solid but two separate entities. In your case PA-CF and PLA must be separate solids anyway, as PA and PLA (similarly to PETG and PLA) do not bond.

SaltWei, 2024-03-22:
[image]

closed 2024-03-22 by SaltWei

ZigZagT, 2024-11-05:
I found the "temperature difference" check was introduced in commit https://github.com/SoftFever/OrcaSlicer/commit/96abc3209a0d4bb8f595cd8473e5796ffa6e3bd1

And it is configurable.

For me personally, I wanted to print TPU & ABS together. So I edited file `C:\Program Files\OrcaSlicer\resources\info\filament_info.json` and added both ABS and TPU to `"high_low_compatible_filament":`. 

This will allow ABS or TPU to be printed together with any filament. The clog risk is real, so use with caution!

zbrozek, 2024-11-22:
It's interesting that this is an explicit compatibility list and not a temperature delta check. Some flavors of PLA and ABS are quite-close in printing temperature and I think there is no risk of clogging. Further, I don't want my slicer to be nannying me anyway. I should be able to either disable this check or dismiss it and proceed anyway.

In fact, I find it kind of annoying that the slicer has any idea of what the polymer is at all. Why?

dgutzler, 2025-05-22:
how to allow this in bambu Studio. It saysthis can be enabled in prefereces

[image]

SaltWei, 2025-05-22:
> how to allow this in bambu Studio. It saysthis can be enabled in prefereces
> 
> [image]

[image]

dgutzler, 2025-05-22:
found it. problem resolved