bambulab/BambuStudio #3744: Can't set 3rd party filament in AMS on Device page.

tonsoffun2325, 2024-03-20:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 PRO

### Printer

X!C

### How to reproduce

On device page,
select filament slot in AMS
Select custom filament from drop down menu.
filament profile and calibrated K value auto populates.
Select submit.
filament shows up in AMS but about 10 seconds later, the filament disappears from AMS.

### Actual results

[image]

This is what it looks like before I select a custom filament, and about 10 seconds after I have selected the custom filament.

### Expected results

The filament should show up in the AMS slot until I change it.

### Project file & Debug log uploads

[New Compressed (zipped) Folder.zip](https://github.com/bambulab/BambuStudio/files/14667836/New.Compressed.zipped.Folder.zip)
[BL LED Riser TPU parts.zip](https://github.com/bambulab/BambuStudio/files/14667842/BL.LED.Riser.TPU.parts.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tonsoffun2325, 2024-03-20:
https://github.com/bambulab/BambuStudio/assets/60978768/f0612ff5-7b8e-4d6a-ab5a-7aeb754c7fdc

DanBao-Bambu, 2024-03-24:
1. Ensure that the user's preset cloud synchronization is enabled
2. X1C model
3. After creating a custom Filament, go to the printer screen and enter the Filament settings interface to check if your custom Filament already exists. This ensures that the printer has been synchronized from the cloud to the newly created custom Filament.
4. Custom Filament can be set up normally on Studio.
I know that setting up the printer is quite complicated, and we are currently optimizing the logic to provide users with a better user experience.
For specific details, please refer to the following link:
https://wiki.bambulab.com/en/software/bambu-studio/custom-filament-issue

DanBao-Bambu, 2024-03-24:
In addition, Studio is about to release a new beta, which does not require operation on the X1C screen, but requires a new software version of X1C to work with.

capsel22, 2024-03-25:
same issue in 1.9 ... wait for it
[image]

DanBao-Bambu, 2024-03-26:
@capsel22 Indeed, to solve this problem, a new printer version is needed in conjunction.

capsel22, 2024-03-26:
> @capsel22 Indeed, to solve this problem, a new printer version is needed in conjunction.

Thanks. Makes sense. It does work when I use third party filament setting on the printer. I guess this will need to do for now :)

closed 2024-10-17 by walterwongbbl