bambulab/BambuStudio #3746: Bambu Studio Turkish Language pack

fatih5228, 2024-03-20:
**Is your feature request related to a problem? Please describe.**
There is no Turkish language file available.

**Describe the solution you'd like**
I want the language package to be added.

**Describe alternatives you've considered**
Can you process the Turkish language translation I provided below?

[BambuStudio_tr.zip](https://github.com/bambulab/BambuStudio/files/14675550/BambuStudio_tr.zip)

fatih5228, 2024-03-23:
Is there anyone who might be interested?

lanewei120, 2024-03-25:
hello @fatih5228 

thanks for doing this
we have a guide here https://wiki.bambulab.com/en/bambu-studio/contribute-languages

could you submit your patch to github directly

fatih5228, 2024-03-26:
File sent. Thanks