bambulab/BambuStudio #3749: Slicer creates invalid color pattern

nowozin, 2024-03-21:
### Bambu Studio Version

1.8.4

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04 x64

### Additional system information

CPU: AMD Ryzen 9 7950X3D
GPU: NVIDIA GeForce RTX 4090, CUDA 12.2, driver 535.154.05

Bambu software version: `BambuStudio_linux_ubuntu_v01.08.04.51_20240117.AppImage`

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Unzip attached `demo.zip` file to obtain `demo.3mf` file.
2. Open Bambu Studio 1.8.4, open the `demo.3mf` file.
3. In the _Prepare_ view, check that there is an all-around golden color in the middle layers (the middle 2.5mm) of the object.  From all sides you can see these layers are colored gold.
4. Slice the object.
5. In the _Preview_ view, now inspect the middle layers which should be solid gold.  You will see they are only about 50% gold, the rest is an invalid white color.
6. (Optional:) I verified that when printed the object is printed as shown in the _Preview_ window, i.e. with the wrong mixed gold/white color in the middle layers.

### Actual results

In the _Preview_, the middle layers contain white and should not:
[image]

### Expected results

In the _Preview_, the middle layers should be solid gold, as shown in the _Prepare_ view:
[image]
[image]

### Project file & Debug log uploads

[demo.zip](https://github.com/bambulab/BambuStudio/files/14707202/demo.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-03-27:
Thanks for reporting this issue, it is a bug, we will try to fix it.

zhimin-zeng-bambulab, 2024-04-16:
It has been fixed in v01.09.00.70, you can have a try.

closed 2024-04-16 by zhimin-zeng-bambulab