bambulab/BambuStudio #3750: Can't edit text

GHTho, 2024-03-21:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

OS/X 14.2.1

### Additional system information

M2 Pro

### Printer

P1S

### How to reproduce

Choose Object right click Edit Text; text box comes up you can change font and text, but after closing the Text box the text remains the same

### Actual results

Text doesn't change

### Expected results

Text should change

### Project file & Debug log uploads

[Archive.zip](https://github.com/bambulab/BambuStudio/files/14707337/Archive.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-03-27:
Thanks for reporting this issue, we will fix it

zhimin-zeng-bambulab, 2024-04-16:
It has been fixed in v01.09.00.70, you can have a try.

closed 2024-04-16 by zhimin-zeng-bambulab