bambulab/BambuStudio #3753: [BUG: Transform/rotate Arrows too small on OSX] BUG: Transform/rotate Arrows too small on OSX

BambuLabDiscord, 2024-03-22:
Hey all - any chance we can get a fix for the arrows being to small on OSX in Bambu Studio? On PC they’re nice and big and easy to grab, but on OSX, they’re only a few pixels across on a high DPI monitor so they’re REALLY small.

kertgartner, 2024-03-22:
[image]
Here's a screenshot that illustrates this issue. A fix would be very welcome, thank you.

Haidiye00, 2024-03-25:
[image]
look this

kertgartner, 2024-03-27:
Yes, I understand this, but the triangular 'grabbers' are very small on OSX compared to windows. when using a high dpi monitor they're VERY hard to grab. Can the size of the triangular grabbers be increased?

Haidiye00, 2024-03-27:
May I ask, what is the resolution of your OSX computer

kertgartner, 2024-03-27:
I'm using an Apple Studio Display, and the resolution is 5120 x 2880.

suchabadperson, 2024-03-27:
I wrote about this one year ago... #1336

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~The size of the gripper is in the preference settings.

closed 2024-10-16 by Haidiye00