bambulab/BambuStudio #3755: New Filament profiles does not register with printer

x348, 2024-03-24:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

windwos it running on a VM, host machine is a AMD ryzon 7

### Printer

Bambu lab x1c

### How to reproduce

1) create a new filament profile
2) change the filamant profile in babu lab studio, for the ams
3) after a short while the profile information changes to ?

i can load the profile from the printer interface, as the only way to access the new profile

### Actual results

unable to use the new profile, untill it's loaded from the printer

### Expected results

this should work the same as the interface on the printer.

### Project file & Debug log uploads

ther is no relevant log or project files

### Checklist of files to include

- [X] Log file
- [X] Project file

DanBao-Bambu, 2024-03-26:
After loading Filament from the printer once, you can successfully set it up from Studio.

DanBao-Bambu, 2024-08-28:
This issue should have been resolved and will be closed. Please feel free to contact us again if you have any further questions.

closed 2024-08-28 by DanBao-Bambu