bambulab/BambuStudio #3756: File causes both axes to drive off the end of their exent

nhnifong, 2024-03-24:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro

### Printer

Bambu Lab A1

### How to reproduce

Slice the attached file for a bambu A1 with 0.8mm nozzle and print.

### Actual results

On about layer 3, the head goes off the right end of the y axis and pushes against the cutter, then it moves in an arc relative to the plate, but not where it should be extruding the part.
It appears to be tracing the motion of layer 3 but mirrored in both axes, and scales outside the extent of the build plate.
An error is produced saying the cutter is stuck, but at this point the printer will make such loud and broken sounding noise that I turn it off so that I don't ruin my belts and motors or something.

### Expected results

Print should be completed normally.

### Project file & Debug log uploads

[filament dryer section test.zip](https://github.com/bambulab/BambuStudio/files/14735429/filament.dryer.section.test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

nhnifong, 2024-03-24:
See the linked video. problem occurs in the middle of layer 4
Tried running a full device calibration but this did not solve it.
https://photos.app.goo.gl/NhvXBySufXyHrqxr7
Firmware 01.02.00.01

nhnifong, 2024-03-25:
I've since printed 2 other things and had no trouble with them, which increases the odds to me that there is something specifically wrong with this gcode.

gaolegao-lx, 2024-08-07:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

nhnifong, 2024-08-07:
This problem is completely solved a couple of months ago by replacing the
USB cable

XunZhangBambu, 2024-08-09:
@nhnifong I'm glad that you have resolved this issue. Will close it now

closed 2024-08-09 by XunZhangBambu