bambulab/BambuStudio #3757: Windows 11 camera in Bambu Studio laggy

cmburns, 2024-03-24:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz   2.80 GHz

### Printer

Bambu X1 carbon

### How to reproduce

1. Open Bambu Studio
2. Load in any object to print
3. Start printing
4. Open camera view
5. See how laggy it is 

### Actual results

The camera is very laggy. It takes long to load the video stream. Then it plays video stream but very laggy. Some sequences are ok but the the stream freezes, then it goes one or two images further but still images no video, then it is smooth for 1 or 2 further seconds then it freezes and so on and so on.
I am in same network as my mobile phone. On my mobile phone I can see the stream nearly in realtime.
So there cannot be any issue with my network since on my mobilde phone it runs smoothly.

### Expected results

Same smooth video stream like in Bamby Handy app
[silicate_box.zip](https://github.com/bambulab/BambuStudio/files/14735764/silicate_box.zip)

### Project file & Debug log uploads

[silicate_box.zip](https://github.com/bambulab/BambuStudio/files/14735766/silicate_box.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-03-25:
Please provider Studio log and a video shown the problem.

jwalker55, 2024-03-26:
I've noticed this problem a lot recently. Camera is fine in Bambu Handy, but sometimes won't even load at all in Bambu Studio. If it does load, it's very unstable with lots of freezes as if there is network congestion. Both devices on the same LAN with gigabit symmetrical internet.

padcom, 2025-07-25:
Guys, I know this is crazy but I've been playing around with this for 10 minutes and it ALWAYS works! Check it out:

- go to the device tab and start playing the video feed
- if it is choppy go to the drop down at the top of the screen and open up Preferences

And as if by magic the feed becomes completely fluid, no chopping whatsoever.