bambulab/BambuStudio #3770: Text in some windows extends beyond window border

VN-Jonathan, 2024-03-25:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.4

### Additional system information

Issue is present on multiple systems. 

### Printer

P1S

### How to reproduce

See attached screenshots for examples. Appears in multiple places. 
[image]
[image]

### Actual results

Text is not visible as it extends beyond the border of the window.

### Expected results

The text should wrap onto the next line to ensure it stays visible within the borders of the window. 

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [X] Project file

DanBao-Bambu, 2024-03-26:
The 1.9Beta version of Studio has been released, and the interface interaction has been updated. Any questions can be raised here.

walterwongbbl, 2024-04-17:
@VN-Jonathan 
We will fix this issue. The translation on the button is indeed too long

closed 2024-10-17 by walterwongbbl