bambulab/BambuStudio #3780: invert colors

valentiy75, 2024-03-27:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro 23h2

### Additional system information

AMD Ryzen 7 4700U with Radeon Graphics 2.00 Ghz
16 gb  ram

### Printer

Bambu Lab P1S

### How to reproduce

In the cut preview, everything looks correct, but it prints in inversion those layers where a different filament color occurs
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Actual results

I want black letters on a white background. But I get an inversion precisely in those layers where a different color is applied. The remaining layers are printed as they appear in the preview.

### Expected results

I expect to see the result as I see in the slicer preview.
[Motorola 3.zip](https://github.com/bambulab/BambuStudio/files/14766746/Motorola.3.zip)

### Project file & Debug log uploads

[Motorola 3.zip](https://github.com/bambulab/BambuStudio/files/14766761/Motorola.3.zip)
[Motorola 3 full.zip](https://github.com/bambulab/BambuStudio/files/14766782/Motorola.3.full.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

valentiy75, 2024-03-27:
[image]

zhimin-zeng-bambulab, 2024-04-02:
[image]
Because your flushing volume is set incorrectly, you could click "Recalculate" to get the correct flush volume.

closed 2024-05-16 by zhimin-zeng-bambulab