bambulab/BambuStudio #3782: 1.9.0 Beta - Behavior of added parts / negative parts coordinates

suchabadperson, 2024-03-27:
### Bambu Studio Version

1.9.0.60

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 17.4.1

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Open Bambu Studio beta version 1.9.0.60
2. Import a model with multiple parts or add a new part to an existing model
3. Select the added part
4. Use the Move tool to reposition the part

### Actual results

When adding a part to a model in the beta version of Bambu Studio and using the Move tool, the origin point for the part's movement is set to the corner of the build plate instead of the center of the main object, which was the behavior in previous versions.
[image]

### Expected results

The origin point should be set to the center of the main object.
[image]

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-03-28:
Can you provide the 3mf project? Thank you

Haidiye00, 2024-03-28:
Why do I feel like there's no problem? Isn't your generic-clue at the world origin?

suchabadperson, 2024-03-28:
> Why do I feel like there's no problem? Isn't your generic-clue at the world origin?

Please see the screenshots above, where you can see the difference between version 1.8.3.89 and the current beta.

Until now, when you added a part to an object (or the object had different parts), its center of coordinates was the parent. Now the center of the added part is the origin of the printer (the lower left part).

Haidiye00, 2024-03-29:
Could you please record a GIF animation? Thank you, you can use ScreenToGif software

Haidiye00, 2024-03-29:
We will add an object coordinate system in the future, which should be the same as your idea. Stay tuned

suchabadperson, 2024-03-29:
> Could you please record a GIF animation? Thank you, you can use ScreenToGif software

Behavior in beta release:
[image]

In 1.8.3.89:
[image]

suchabadperson, 2024-04-08:
Issue still present in 1.90 beta2.

Haidiye00, 2024-04-09:
Sorry, the code is still under development

suchabadperson, 2024-05-13:
Issue still present in 1.9.1 Public Official release.

Haidiye00, 2024-05-14:
[image]
Sorry for keeping you waiting. This feature has been developed and is currently in beta testing. The next version will be released. Stay tuned.

ReneZeidler, 2024-06-17:
> This feature has been developed and is currently in internal testing. The next version will be released. Stay tuned.

1.9.2 has been released last week and this feature isn't included. Is this still being worked on?

Haidiye00, 2024-06-18:
Sorry, it doesn't include

suchabadperson, 2024-08-26:
The same issue is present in 1.9.4. This doesn't occur in OrcaSlicer (where the center of coordinates for an added part aligns with the main object) or in Bambu Studio versions before 1.9.0.

Haidiye00, 2024-08-27:
Sorry, this feature is still in beta testing, please wait~

suchabadperson, 2024-10-15:
Object coordinates appear to have been added in the latest v01.10.00.74 - 1.10.0 Public Beta. Well done!”

Haidiye00, 2024-10-16:
Thank you, we will continue to work hard~

closed 2024-10-16 by Haidiye00