bambulab/BambuStudio #3792: Buttons on 'Prepare' screen are invisible

lesserevil, 2024-03-29:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

VirtualBox VM running on Ubuntu 22.04

### Printer

X1C

### How to reproduce

Open app, go to 'preview' screen, 

### Actual results

No buttons along the top. If I mouse over where the buttons should be, I can see the tooltips, and I can click on them but I can't see them.

### Expected results

Buttons visible

### Project file & Debug log uploads

Happens on all projects, even empty ones.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-04-01:
we have linux versions, could you try linux version instead of the windows version on virtual box?

lesserevil, 2024-06-04:
Moved to the linux version, no longer having the issue.

closed 2024-06-04 by lesserevil