bambulab/BambuStudio #3793: Display flow value when using "Flow Dynamic Calibration" before print (X1C)

capsel22, 2024-03-29:
**Is your feature request related to a problem? Please describe.**
When sending a print, there is an option to do Flow Dynamic Calibration.
It works as intended, however the calibrated value is not shown. It would be quite useful this value was displayed so we can update the filament settings with that value.
[image]

**Describe the solution you'd like**
Display flow value after performing Flow Dynamic Calibration in send print window

**Describe alternatives you've considered**
To get the value, currently you need to run Automatic or Manual flow calibration in calibration tab.

Sabotin, 2024-03-30:
I suspect you're aware that this feature does not calibrate the **Flow Rate** but only a **K factor** _a.k.a pressure advance_.

For flow rate, you'll gotta go to the calibrate tab and calibrate there.

capsel22, 2024-03-30:
> I suspect you're aware that this feature does not calibrate the **Flow Rate** but only a **K factor** _a.k.a pressure advance_.
> 
> For flow rate, you'll gotta go to the calibrate tab and calibrate there.

thanks. I clarified in my original post.

beto-optiart, 2026-08-18:
upvote