bambulab/BambuStudio #3794: Mac M1: Studio crashes after sending to print to Cloud

MacMarc, 2024-03-29:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.4.1

### Additional system information

MacBook Air M1

### Printer

BambuLb P1S

### How to reproduce

1. create project
2. slice project
3. print print
4. open an other app like firefox

### Actual results

bambustudio crashed in the background

### Expected results

bambustudio doesn't crash

### Project file & Debug log uploads

[crashreport.txt.zip](https://github.com/bambulab/BambuStudio/files/14801474/crashreport.txt.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

DomPerignon3, 2024-03-30:
Same issue on MBPro M2, but it’s doesn’t crash all the time. Might/not be related but my A1 mini told me to reformat the original SD card, which had filsesys problems. I used MacOS Disk utility’s SOS to repair it, and no BS crashing since them.

DomPerignon3, 2024-03-31:
Still happening with Bambu Studio 1.9.0.60.

rekonnected, 2024-04-05:
Same crash here on an M1Pro MBP, Sonoma 14.3 (23D56). Fairly reliably crashes immediately after the first upload after opening the app, though it is worth noting the upload goes through and the print successfully initiates on the printer (P1S). When reopening, the app does not crash as consistently, but it does still occasionally crash on upload.

BambuStudio Beta Version: 1.9.0.60
Network Plugin Version: 01.09.00.03

DomPerignon3, 2024-04-09:
Yup, it's reliably frequent. I've had it crash when I switched to a second window of BBS that was already opened, so it's not linked to a specific app. I have the crash report if anyone from BBL is looking into it.

mboman00, 2024-04-24:
Had a crash this morning where the restore didn't work.  The print went to the printer (A1 mini) though.

Bambu Studio Beta 1.9.0.70

DomPerignon3, 2024-06-21:
Hello,

Here is a first one:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [16610]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.09.02.57 ()
Code Type:             ARM-64 (Native)
Parent Process:        BambuStudio [16447]
User ID:               501

Date/Time:             2024-06-20 21:26:04.7234 -0400
OS Version:            macOS 14.4 (23E214)
Report Version:        12
Anonymous UUID:        E87B8067-77B1-18C3-CE31-1A3BEE0C4DE0

Sleep/Wake UUID:       7B28107B-F4FB-4876-ACD6-FEE36E9ABAFD

Time Awake Since Boot: 170000 seconds
Time Since Wake:       14726 seconds

System Integrity Protection: enabled

Crashed Thread:        28

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0xfffffffffffffff0
Exception Codes:       0x0000000000000001, 0xfffffffffffffff0

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [16610]

VM Region Info: 0xfffffffffffffff0 is not in any region.
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->
      UNUSED SPACE AT END

Thread 0:: com.apple.NSEventThread
0   libsystem_kernel.dylib                 0x1857461f4 mach_msg2_trap + 8
1   libsystem_kernel.dylib                 0x185758b24 mach_msg2_internal + 80
2   libsystem_kernel.dylib                 0x18574ee34 mach_msg_overwrite + 476
3   libsystem_kernel.dylib                 0x185746578 mach_msg + 24
4   CoreFoundation                         0x185866058 __CFRunLoopServiceMachPort + 160
5   CoreFoundation                         0x18586491c __CFRunLoopRun + 1208
6   CoreFoundation                         0x185863e0c CFRunLoopRunSpecific + 608
7   AppKit                                 0x1891e5cb4 _NSEventThread + 144
8   libsystem_pthread.dylib                0x185786f94 _pthread_start + 136
9   libsystem_pthread.dylib                0x185781d34 thread_start + 8

Thread 1:
0   libsystem_kernel.dylib                 0x1857499ec __psynch_cvwait + 8
1   libsystem_pthread.dylib                0x18578755c _pthread_cond_wait + 1228
2   libc++.1.dylib                         0x1856acb14 std::__1::condition_variable::wait(std::__1::unique_lock<std::__1::mutex>&) + 28
3   libbambu_networking.dylib              0x119a155c0 BBL::TunnelMqtt::reconnect(std::__1::unique_lock<std::__1::mutex>&, int) + 696
4   libbambu_networking.dylib              0x119a150d0 BBL::TunnelMqtt::run() + 828
5   libbambu_networking.dylib              0x119a18958 0x119900000 + 1149272
6   libsystem_pthread.dylib                0x185786f94 _pthread_start + 136
7   libsystem_pthread.dylib                0x185781d34 thread_start + 8

Thread 2:: JavaScriptCore libpas scavenger
0   libsystem_kernel.dylib                 0x1857499ec __psynch_cvwait + 8
1   libsystem_pthread.dylib                0x18578755c _pthread_cond_wait + 1228
2   JavaScriptCore                         0x1a310417c scavenger_thread_main + 1512
3   libsystem_pthread.dylib                0x185786f94 _pthread_start + 136
[... 345 lines cut ...]
  "procName" : "BambuStudio",
  "procPath" : "\/Applications\/BambuStudio.app\/Contents\/MacOS\/BambuStudio",
  "bundleInfo" : {"CFBundleShortVersionString":"01.09.02.57","CFBundleVersion":"","CFBundleIdentifier":"com.bambulab.bambu-studio"},
  "storeInfo" : {"deviceIdentifierForVendor":"D7D96D39-8E0D-528D-B441-1279CECC3CF7","thirdParty":true},
  "parentProc" : "BambuStudio",
  "parentPid" : 16447,
  "coalitionName" : "com.bambulab.bambu-studio",
  "crashReporterKey" : "E87B8067-77B1-18C3-CE31-1A3BEE0C4DE0",
  "responsiblePid" : 47950,
  "codeSigningID" : "com.bambulab.bambu-studio",
  "codeSigningTeamID" : "T3UBR9Y3B2",
  "codeSigningFlags" : 570495745,
  "codeSigningValidationCategory" : 6,
  "codeSigningTrustLevel" : 4294967295,
  "instructionByteStream" : {"beforePC":"\/w9f1pPu\/5d\/IwPV9E++qf17Aan9QwCREABA+fEDAKoxXO3yMBrB2g==","atPC":"CA5f+OkDEKr0AwCq8QMJqvF39\/IRCT\/X\/XtBqfRPwqj\/D1\/WEABA+Q=="},
  "wakeTime" : 14726,
  "sleepWakeUUID" : "7B28107B-F4FB-4876-ACD6-FEE36E9ABAFD",
  "sip" : "enabled",
  "vmRegionInfo" : "0xfffffffffffffff0 is not in any region.  \n      REGION TYPE                    START - END         [ VSIZE] PRT\/MAX SHRMOD  REGION DETAIL\n      UNUSED SPACE AT START\n--->  \n      UNUSED SPACE AT END",
  "exception" : {"codes":"0x0000000000000001, 0xfffffffffffffff0","rawCodes":[1,-16],"type":"EXC_BAD_ACCESS","signal":"SIGSEGV","subtype":"KERN_INVALID_ADDRESS at 0xfffffffffffffff0"},
  "termination" : {"flags":0,"code":11,"namespace":"SIGNAL","indicator":"Segmentation fault: 11","byProc":"exc handler","byPid":16610},
  "vmregioninfo" : "0xfffffffffffffff0 is not in any region.  \n      REGION TYPE                    START - END         [ VSIZE] PRT\/MAX SHRMOD  REGION DETAIL\n      UNUSED SPACE AT START\n--->  \n      UNUSED SPACE AT END",
  "extMods" : {"caller":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"system":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"targeted":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"warnings":0},
  "faultingThread" : 28,
[... cut ...] cation":136,"imageIndex":30},{"imageOffset":7476,"symbol":"thread_start","symbolLocation":8,"imageIndex":30}]},{"id":2838574,"frames":[{"imageOffset":18924,"symbol":"__psynch_cvwait","symbolLocation":8,"imageIndex":27},{"imageOffset":30044,"symbol":"_pthread_cond_wait","symbolLocation":1228,"imageIndex":30},{"imageOffset":68372,"symbol":"std::__1::condition_variable::wait(std::__1::unique_lock<std::__1::mutex>&)","symbolLocation":28,"imageIndex":32},{"imageOffset":88040,"symbol":"CloseTaskQueue:

DomPerignon3, 2024-06-21:
Here is second crash:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [13056]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.09.00.60 ()
Code Type:             ARM-64 (Native)
Parent Process:        BambuStudio [10501]
User ID:               501

Date/Time:             2024-04-08 20:41:19.4930 -0400
OS Version:            macOS 14.4 (23E214)
Report Version:        12
Anonymous UUID:        E87B8067-77B1-18C3-CE31-1A3BEE0C4DE0

Sleep/Wake UUID:       60375EAD-F274-437F-8FBA-3B55667A1A9D

Time Awake Since Boot: 240000 seconds
Time Since Wake:       1011 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00004c8e806d99a0
Exception Codes:       0x0000000000000001, 0x00004c8e806d99a0

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [13056]

VM Region Info: 0x4c8e806d99a0 is not in any region.  Bytes after previous region: 83693887396257  Bytes before following region: 21378192533088
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  reserved VM address space (unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600020000000 [512.0M] rw-/rwx SM=PRV

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib                        0x1817306b4 objc_release + 16
1   AppKit                                 0x1858079ec 0x185402000 + 4217324
2   AppKit                                 0x1857eed4c 0x185402000 + 4115788
3   AppKit                                 0x185a8758c -[NSViewBackingLayer setNeedsDisplayInRect:] + 176
4   AppKit                                 0x1854549dc -[NSView setNeedsDisplayInRect:] + 756
5   AppKit                                 0x18546ff68 -[NSTextField updateCell:] + 80
6   AppKit                                 0x18546f7f0 -[NSTextFieldCell setTextColor:] + 204
7   BambuStudio                            0x103a101c4 -[wxNSTextField setEnabled:] + 100
8   BambuStudio                            0x103a1d9d8 wxWidgetCocoaImpl::Enable(bool) + 128
9   BambuStudio                            0x103997f54 wxWindow::DoEnable(bool) + 48
10  BambuStudio                            0x1038e97e4 wxWindowBase::NotifyWindowOnEnableChange(bool) + 36
11  BambuStudio                            0x1038e98b4 wxWindowBase::Enable(bool) + 88
12  BambuStudio                            0x1016fca48 TextInput::Enable(bool) + 44
13  BambuStudio                            0x101e547f4 Slic3r::GUI::SelectMachineDialog::show_status(Slic3r::GUI::PrintDialogStatus, std::__1::vector<wxString, std::__1::allocator<wxString>>) + 492
14  BambuStudio                            0x101e6270c Slic3r::GUI::SelectMachineDialog::Show(bool) + 48
15  BambuStudio                            0x10398f4f0 wxDialog::EndModal(int) + 68
16  BambuStudio                            0x101a5bad0 Slic3r::GUI::Plater::print_job_finished(wxCommandEvent&) + 140
17  BambuStudio                            0x103ad0778 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
18  BambuStudio                            0x103ad0458 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
19  BambuStudio                            0x103ad027c wxEvtHandler::ProcessEvent(wxEvent&) + 56
20  BambuStudio                            0x103ad00cc wxEvtHandler::ProcessPendingEvents() + 236
21  BambuStudio                            0x103a33f34 wxAppConsoleBase::ProcessPendingEvents() + 112
22  BambuStudio                            0x103ad8f88 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 60
23  CoreFoundation                         0x181be5254 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
24  CoreFoundation                         0x181be5140 __CFRunLoopDoObservers + 536
25  CoreFoundation                         0x181be476c __CFRunLoopRun + 776
26  CoreFoundation                         0x181be3e0c CFRunLoopRunSpecific + 608
27  HIToolbox                              0x18c37f000 RunCurrentEventLoopInMode + 292
[... 382 lines cut ...]
  "procName" : "BambuStudio",
  "procPath" : "\/Applications\/BambuStudio.app\/Contents\/MacOS\/BambuStudio",
  "bundleInfo" : {"CFBundleShortVersionString":"01.09.00.60","CFBundleVersion":"","CFBundleIdentifier":"com.bambulab.bambu-studio"},
  "storeInfo" : {"deviceIdentifierForVendor":"21C4276E-321F-561C-9F3C-D35CC09141F8","thirdParty":true},
  "parentProc" : "BambuStudio",
  "parentPid" : 10501,
  "coalitionName" : "com.bambulab.bambu-studio",
  "crashReporterKey" : "E87B8067-77B1-18C3-CE31-1A3BEE0C4DE0",
  "responsiblePid" : 95377,
  "codeSigningID" : "com.bambulab.bambu-studio",
  "codeSigningTeamID" : "F4SKHPXDD9",
  "codeSigningFlags" : 570495745,
  "codeSigningValidationCategory" : 6,
  "codeSigningTrustLevel" : 4294967295,
  "instructionByteStream" : {"beforePC":"gf7\/VMADX9bAA1\/WYUMw0CFIHJFYBQAUAAAA6m3\/\/1QQAED5Aq59kg==","atPC":"URBA+TEDEDbwAwA2Ef5305H+\/7Q\/BgDxYAIAVBEg4NIRAhHL4QMQqg=="},
  "wakeTime" : 1011,
[... cut ...] imageOffset":43272252,"symbol":"tbb::detail::r1::rml::private_worker::run()","symbolLocation":412,"imageIndex":26},{"imageOffset":43271828,"symbol":"tbb::detail::r1::rml::private_worker::thread_routine(void*)","symbolLocation":12,"imageIndex":26},{"imageOffset":28564,"symbol":"_pthread_start","symbolLocation":136,"imageIndex":34},{"imageOffset":7476,"symbol":"thread_start","symbolLocation":8,"imageIndex":34}]},{"id":3681815,"name":"slic3r_tbb_9","threadState":{"x":[{"value":14},{"value":0},{"val

DomPerignon3, 2024-06-22:
Hello,

here's the most recent one.

Let me know if you need more.

m

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [30263]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.09.02.57 ()
Code Type:             ARM-64 (Native)
Parent Process:        BambuStudio [49357]
User ID:               501

Date/Time:             2024-06-22 00:15:16.1413 -0400
OS Version:            macOS 14.4 (23E214)
Report Version:        12
Anonymous UUID:        E87B8067-77B1-18C3-CE31-1A3BEE0C4DE0

Sleep/Wake UUID:       1DBE18FE-D9DE-4709-A3B9-C3806CABC208

Time Awake Since Boot: 200000 seconds
Time Since Wake:       890 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   BambuStudio [30263]

Application Specific Information:
abort() called

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib                 0x18574ea60 __pthread_kill + 8
1   libsystem_pthread.dylib                0x185786c20 pthread_kill + 288
2   libsystem_c.dylib                      0x185693a20 abort + 180
3   libc++abi.dylib                        0x18573dd30 abort_message + 132
4   libc++abi.dylib                        0x18572dfe8 demangling_terminate_handler() + 348
5   libobjc.A.dylib                        0x1853cc1d0 _objc_terminate() + 144
6   libc++abi.dylib                        0x18573d0f4 std::__terminate(void (*)()) + 16
7   libc++abi.dylib                        0x1857405f8 __cxa_rethrow + 204
8   libobjc.A.dylib                        0x1853e62d4 objc_exception_rethrow + 44
9   AppKit                                 0x189346c40 -[NSCoreDragManager dragImage:fromWindow:at:offset:event:pasteboard:source:slideBack:] + 1664
10  WebKit                                 0x1a92fe678 WebKit::WebViewImpl::startDrag(WebCore::DragItem const&, WebKit::ShareableBitmapHandle&&) + 620
11  WebKit                                 0x1a987e170 WebKit::WebPageProxy::didReceiveMessage(IPC::Connection&, IPC::Decoder&) + 24936
12  WebKit                                 0x1a9946b6c IPC::MessageReceiverMap::dispatchMessage(IPC::Connection&, IPC::Decoder&) + 264
13  WebKit                                 0x1a9438938 WebKit::WebProcessProxy::didReceiveMessage(IPC::Connection&, IPC::Decoder&) + 44
14  WebKit                                 0x1a9941c48 IPC::Connection::dispatchMessage(std::__1::unique_ptr<IPC::Decoder, std::__1::default_delete<IPC::Decoder>>) + 324
15  WebKit                                 0x1a99420d0 IPC::Connection::dispatchIncomingMessages() + 556
16  JavaScriptCore                         0x1a1d0ffa0 WTF::RunLoop::performWork() + 204
17  JavaScriptCore                         0x1a1d10ec8 WTF::RunLoop::performWork(void*) + 36
18  CoreFoundation                         0x185865eb0 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 28
19  CoreFoundation                         0x185865e44 __CFRunLoopDoSource0 + 176
20  CoreFoundation                         0x185865bb4 __CFRunLoopDoSources0 + 244
21  CoreFoundation                         0x1858647a0 __CFRunLoopRun + 828
22  CoreFoundation                         0x185863e0c CFRunLoopRunSpecific + 608
23  HIToolbox                              0x18ffff000 RunCurrentEventLoopInMode + 292
24  HIToolbox                              0x18fffee3c ReceiveNextEventCommon + 648
25  HIToolbox                              0x18fffeb94 _BlockUntilNextEventMatchingListInModeWithFilter + 76
[... 385 lines cut ...]
  "procName" : "BambuStudio",
  "procPath" : "\/Applications\/BambuStudio.app\/Contents\/MacOS\/BambuStudio",
  "bundleInfo" : {"CFBundleShortVersionString":"01.09.02.57","CFBundleVersion":"","CFBundleIdentifier":"com.bambulab.bambu-studio"},
  "storeInfo" : {"deviceIdentifierForVendor":"D7D96D39-8E0D-528D-B441-1279CECC3CF7","thirdParty":true},
  "parentProc" : "BambuStudio",
  "parentPid" : 49357,
  "coalitionName" : "com.bambulab.bambu-studio",
  "crashReporterKey" : "E87B8067-77B1-18C3-CE31-1A3BEE0C4DE0",
  "responsiblePid" : 47950,
  "codeSigningID" : "com.bambulab.bambu-studio",
  "codeSigningTeamID" : "T3UBR9Y3B2",
  "codeSigningFlags" : 570495745,
  "codeSigningValidationCategory" : 6,
  "codeSigningTrustLevel" : 4294967295,
  "instructionByteStream" : {"beforePC":"fyMD1f17v6n9AwCRW+D\/l78DAJH9e8Go\/w9f1sADX9YQKYDSARAA1A==","atPC":"AwEAVH8jA9X9e7+p\/QMAkVDg\/5e\/AwCR\/XvBqP8PX9bAA1\/WcAqA0g=="},
  "wakeTime" : 890,
  "sleepWakeUUID" : "1DBE18FE-D9DE-4709-A3B9-C3806CABC208",
  "sip" : "enabled",
  "exception" : {"codes":"0x0000000000000000, 0x0000000000000000","rawCodes":[0,0],"type":"EXC_CRASH","signal":"SIGABRT"},
  "termination" : {"flags":0,"code":6,"namespace":"SIGNAL","indicator":"Abort trap: 6","byProc":"BambuStudio","byPid":30263},
  "asi" : {"libsystem_c.dylib":["abort() called"]},
  "extMods" : {"caller":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"system":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"targeted":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"warnings":0},
[... cut ...] 6,"symbol":"tbb::detail::r1::rml::private_worker::run()","symbolLocation":412,"imageIndex":26},{"imageOffset":45912352,"symbol":"tbb::detail::r1::rml::private_worker::thread_routine(void*)","symbolLocation":12,"imageIndex":26},{"imageOffset":28564,"symbol":"_pthread_start","symbolLocation":136,"imageIndex":28},{"imageOffset":7476,"symbol":"thread_start","symbolLocation":8,"imageIndex":28}],"threadState":{"x":[{"value":14},{"value":0},{"value":0},{"value":1},{"value":105553178693696},{"value":0},

bambu123, 2024-06-24:
@DomPerignon3 Did they all crash when sending the print?

DomPerignon3, 2024-06-26:
Happened on an Intel MacBook as well.

DomPerignon3, 2024-07-14:
Here we go again.

===
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [82316]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.09.02.57 ()
Code Type:             ARM-64 (Native)
Parent Process:        BambuStudio [61999]
User ID:               501

Date/Time:             2024-07-14 04:28:12.1773 -0400
OS Version:            macOS 14.5 (23F79)
Report Version:        12
Anonymous UUID:        E87B8067-77B1-18C3-CE31-1A3BEE0C4DE0

Sleep/Wake UUID:       F86F9F80-6B05-4A3B-86A2-FBB48D6B6227

Time Awake Since Boot: 150000 seconds
Time Since Wake:       37035 seconds

System Integrity Protection: enabled

Crashed Thread:        33

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x000000019d653ae4

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [82316]

Application Specific Information:
abort() called

Thread 0:: bambustu_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x19d707fac __ulock_wait + 8
1   libsystem_pthread.dylib       	       0x19d74948c _pthread_join + 608
2   BambuStudio                   	       0x103d2f860 boost::thread::join_noexcept() + 188
3   BambuStudio                   	       0x10399f380 Slic3r::GUI::Job::join(int) + 116
4   BambuStudio                   	       0x103a87e40 Slic3r::GUI::SelectMachineDialog::prepare_mode(bool) + 384
5   BambuStudio                   	       0x103a8d3dc Slic3r::GUI::SelectMachineDialog::show_status(Slic3r::GUI::PrintDialogStatus, std::__1::vector<wxString, std::__1::allocator<wxString>>) + 7836
6   BambuStudio                   	       0x103a9abac Slic3r::GUI::SelectMachineDialog::Show(bool) + 48
7   BambuStudio                   	       0x1057fb008 wxDialog::EndModal(int) + 68
8   BambuStudio                   	       0x10366fd6c Slic3r::GUI::Plater::print_job_finished(wxCommandEvent&) + 140
9   BambuStudio                   	       0x10593bcd4 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
10  BambuStudio                   	       0x10593b9b4 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
11  BambuStudio                   	       0x10593b7d8 wxEvtHandler::ProcessEvent(wxEvent&) + 56
12  BambuStudio                   	       0x10593b628 wxEvtHandler::ProcessPendingEvents() + 236
13  BambuStudio                   	       0x10589f9d0 wxAppConsoleBase::ProcessPendingEvents() + 112
14  BambuStudio                   	       0x1059444e4 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 60
15  CoreFoundation                	       0x19d82587c __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
16  CoreFoundation                	       0x19d825768 __CFRunLoopDoObservers + 536
17  CoreFoundation                	       0x19d824d94 __CFRunLoopRun + 776
18  CoreFoundation                	       0x19d824434 CFRunLoopRunSpecific + 608
19  HIToolbox                     	       0x1a7fc819c RunCurrentEventLoopInMode + 292
20  HIToolbox                     	       0x1a7fc7e54 ReceiveNextEventCommon + 260
21  HIToolbox                     	       0x1a7fc7d30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
22  AppKit                        	       0x1a1083d68 _DPSNextEvent + 660
23  AppKit                        	       0x1a1879808 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
24  BambuStudio                   	       0x10586bd10 wxGUIEventLoop::DoDispatchTimeout(unsigned long) + 180
25  BambuStudio                   	       0x1059449ec wxCFEventLoop::DispatchTimeout(unsigned long) + 44
26  BambuStudio                   	       0x105944b0c wxCFEventLoop::OSXDoRun() + 172
27  BambuStudio                   	       0x10586c1e0 wxModalEventLoop::OSXDoRun() + 88
28  BambuStudio                   	       0x105944bc8 wxCFEventLoop::DoRun() + 40
29  BambuStudio                   	       0x1058c4274 wxEventLoopBase::Run() + 132
30  BambuStudio                   	       0x1057faee4 wxDialog::ShowModal() + 100
[... 660 lines cut ...]
      },
      "deploymentId" : 240000002
    },
    {
      "rolloutId" : "639124e81d92412bfb4880b3",
      "factorPackIds" : {

      },
      "deploymentId" : 240000012
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "c47ab4cc-c9c3-4b5d-a87c-e2433ce02597",
      "experimentId" : "6639bc6ba73d460582162323",
      "deploymentId" : 400000006
    },
    {
      "treatmentId" : "45f4e2a5-551b-4bc2-a2dc-19c244dda8f8",
      "experimentId" : "6643969b3099cf28e049862f",
      "deploymentId" : 400000006
    }
  ]
}
}

DomPerignon3, 2024-07-14:
This is one is **different**: it crashed as I switched from the Prepare to Preview tab, **NOT** after I sent the job to the printer.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [86957]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.09.02.57 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-07-14 19:15:10.4502 -0400
OS Version:            macOS 14.5 (23F79)
Report Version:        12
Anonymous UUID:        E87B8067-77B1-18C3-CE31-1A3BEE0C4DE0

Sleep/Wake UUID:       ACD7F66A-EB80-4C05-9990-D6440B229D76

Time Awake Since Boot: 200000 seconds
Time Since Wake:       90254 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x000000016ae63f00
Exception Codes:       0x0000000000000002, 0x000000016ae63f00

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [86957]

VM Region Info: 0x16ae63f00 is in 0x167660000-0x16ae64000;  bytes after start: 58736384  bytes before end: 255
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      mapped file                 16765c000-167660000    [   16K] r--/rwx SM=COW  Object_id=7bf6c57c
--->  STACK GUARD                 167660000-16ae64000    [ 56.0M] ---/rwx SM=NUL  stack guard for thread 0
      Stack                       16ae64000-16b660000    [ 8176K] rw-/rwx SM=SHM  thread 0

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   CoreFoundation                	       0x19d7b31c8 CFStringFindWithOptionsAndLocale + 36
1   CoreFoundation                	       0x19d7c8a0c CFStringFind + 92
2   CoreText                      	       0x19fbe8594 TTenuousComponentFont::CopyAttribute(unsigned long) const + 1948
3   CoreText                      	       0x19fce96d0 TComponentFont::AddDescriptorAttributes(TInlineVector<__CFString const*, 30ul>*, TInlineVector<void const*, 30ul>*) const + 1128
4   CoreText                      	       0x19fcea234 TTenuousComponentFont::AddDescriptorAttributes(TInlineVector<__CFString const*, 30ul>*, TInlineVector<void const*, 30ul>*) const + 44
5   CoreText                      	       0x19fbdd5ec TBaseFont::CreateDescriptorAttributes() const + 1220
6   CoreText                      	       0x19fbdd0ec TDescriptor::GetAttributes(bool) const + 96
7   CoreText                      	       0x19fbdd060 TDescriptor::CopyAttributes() const + 32
8   CoreText                      	       0x19fbdd018 CTFontDescriptorCopyAttributes + 40
9   CoreText                      	       0x19fbdef24 TFont::CopyDescriptor() const + 112
10  CoreText                      	       0x19fbdee14 CTFontCopyFontDescriptor + 40
11  BambuStudio                   	       0x10770d1c0 wxNativeFontInfo::InitFromFont(__CTFont const*) + 148
12  BambuStudio                   	       0x10770cf70 wxFontRefData::wxFontRefData(__CTFont const*) + 176
13  BambuStudio                   	       0x10770eb24 wxFont::wxFont(wxOSXSystemFont) + 100
14  BambuStudio                   	       0x1051d1174 Label::sysFont(int, bool, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>) + 308
15  BambuStudio                   	       0x105b9093c -[NSTableHeaderCell(Font) font] + 44
16  AppKit                        	       0x1a112994c -[NSCell _effectiveFont] + 56
17  AppKit                        	       0x1a154aaf0 -[NSCell _textAttributesWithInteriorBackgroundStyle:] + 64
18  AppKit                        	       0x1a10b9078 -[NSTextFieldCell _textAttributes] + 140
19  AppKit                        	       0x1a10bbb4c __NSGetStringAndAttributesFromTextCell + 448
20  AppKit                        	       0x1a10bb914 ___NSGetTextCellBoundingRect_block_invoke + 48
21  AppKit                        	       0x1a1073ac8 +[NSAppearance _performWithCurrentAppearance:usingBlock:] + 72
22  AppKit                        	       0x1a10bb874 _NSGetTextCellBoundingRect + 264
23  AppKit                        	       0x1a10bb570 -[NSCell cellSizeForBounds:] + 188
24  AppKit                        	       0x1a10bb34c -[NSTextFieldCell cellSizeForBounds:] + 108
25  AppKit                        	       0x1a18bfa10 -[NSTableHeaderCell cellSizeForBounds:] + 80
26  BambuStudio                   	       0x10775c300 wxCocoaDataViewControl::FitColumnWidthToContent(unsigned int) + 160
27  BambuStudio                   	       0x107766604 wxDataViewCtrl::DoExpand(wxDataViewItem const&, bool) + 172
[... 980 lines cut ...]
    "CFBundleShortVersionString" : "6.9",
    "CFBundleIdentifier" : "com.apple.Foundation",
    "size" : 12967936,
    "uuid" : "99e0292d-7873-3968-9c9c-5955638689a5",
    "path" : "\/System\/Library\/Frameworks\/Foundation.framework\/Versions\/C\/Foundation",
    "name" : "Foundation",
    "CFBundleVersion" : "2503.1"
  },
  {
    "size" : 0,
    "source" : "A",
    "base" : 0,
    "uuid" : "00000000-0000-0000-0000-000000000000"
  },
  {
    "source" : "P",
    "arch" : "arm64e",
    "base" : 6936350720,
    "size" : 241660,
    "uuid" : "9b8b53f9-e2b6-36df-98e9-28d8fca732f2",
    "path" : "\/usr\/lib\/system\/libsystem_kernel.dylib",
    "name" : "libsystem_kernel.dylib"
  },
  {
    "source" : "P",

DomPerignon3, 2024-08-03:
Hello,

in case you need some comparison, here's a random crash (nothing to do with printing, I was just adding some negative volumes to a part).

[BambuStudio-2024-08-02-235322.ips.txt](https://github.com/user-attachments/files/16477983/BambuStudio-2024-08-02-235322.ips.txt)

Thanks.

walterwongbbl, 2024-08-08:
@DomPerignon3 
Thank you.

walterwongbbl, 2024-08-08:
@DomPerignon3 
It seems to be caused by the UI framework we are using.
Is this kind of crash very frequent.

DomPerignon3, 2024-08-10:
On my M2 MBPro I can launch 8 prints without a crash, then get 3 crashes in a row. I don't see a pattern yet.

Also Happens on an older Intel MBAir, though I don't launch prints as often from that one.

Thanks.

Téléchargez Outlook pour iOS<https://aka.ms/o0ukef>

DomPerignon3, 2024-08-15:
Once again, one crash right after sending print (I was using BBS when it occured).  Before that 7 prints without crashes over a couple of days.

[BambuStudio-2024-08-15-120601.ips.zip](https://github.com/user-attachments/files/16630134/BambuStudio-2024-08-15-120601.ips.zip)

DomPerignon3, 2024-08-15:
Second crash in a row. Had rebooted between this one and previous one.
[BambuStudio-2024-08-15-170355.ips.zip](https://github.com/user-attachments/files/16630376/BambuStudio-2024-08-15-170355.ips.zip)

DomPerignon3, 2024-08-31:
BTW that crash issue also prevents users from giving creators Star for their Model and their Print profiles.

I can NOT properly review over 100 greats designs that I printed due to this issue. 

Multiply this by all the MacOS users and you get a lot of creators not getting appropriate ratings.

walterwongbbl, 2024-09-04:
Hi. @DomPerignon3 
Very strange, we have 3 different models of MacBook using version 1.46.1. After conducting various tests, no crashes occurred Only once did a crash occur after opening liveview. We have also seen similar issues in other posts. But we haven't found the pattern yet.

CrossedRifles, 2024-09-04:
I get regular crashes on Mac Studio with M1 chip. They are always directly after sending a job to the printer. Happens about everyday.
[Bambu Studio Crash 1.zip](https://github.com/user-attachments/files/16880504/Bambu.Studio.Crash.1.zip)

DomPerignon3, 2024-10-16:
Hello,

here are a bunch of crash logs. Most are from crashes right after sending, some not.

[Archive.zip](https://github.com/user-attachments/files/17398302/Archive.zip)

DomPerignon3, 2024-11-11:
Hello,
still getting a LOT of crashes, even with 1.9.7.52.

See attached.
[8 crashes 6 days.zip](https://github.com/user-attachments/files/17694686/8.crashes.6.days.zip)

Thanks.

BlaizeDaddi, 2024-12-24:
Mines crashed when I hit print on a large cut file. Annoying...Attached is the zip
[BambuCrash.zip](https://github.com/user-attachments/files/18241441/BambuCrash.zip)