bambulab/BambuStudio #3801: Tree support generation freezes and crashes on donut shapes

Reimajo, 2024-03-31:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro 64-bit (10.0, Build 19045)

### Additional system information

Card name: NVIDIA GeForce RTX 3090
Processor: Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz (12 CPUs), ~3.7GHz
Memory: 32768MB RAM

### Printer

X1C

### How to reproduce

1. Open project (DonutShape.3mf)
2. Change the support type to "Tree (auto)"
3. Start slicing

### Actual results

App freezes around 75% and doesn't continue slicing

### Expected results

Slicing finishes without problems

### Project file & Debug log uploads

[TreeSuportBug.zip](https://github.com/bambulab/BambuStudio/files/14815091/TreeSuportBug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Reimajo, 2024-03-31:
Someone reported this issue in July 2023 [here](https://forum.bambulab.com/t/slicer-freezes-when-using-tree-support/19083) but no bug report has been posted on GitHub by them. They also had the issue with donut-shaped objects, so I believe this could be related. I ran into the same issue, found the forum thread and decided to post my own project file with the same issue here.

tangketan, 2024-04-01:
Sorry but I can't reproduce the bug. Beside, freezing at 75% means it's "Optimizing toolpaths", so nothing to do with tree support.

XunZhangBambu, 2024-04-07:
@Reimajo  In 1.9 beta, slicing this model with tree support does not cause freezing