bambulab/BambuStudio #3805: Unable to enter decimal values

Bokbacken, 2024-04-01:
### Bambu Studio Version

1.9.0.60

### Where is the application from?

Bambu Lab github releases

### OS version

WIndows 11 22.H.2

### Additional system information

Region Sweden
Decimal symbol , (this is probably the problem and this is the Swedish setting)

### Printer

Bambu Labs X1C

### How to reproduce

1. Press "Rensnings volymer" (Flushing volumes)
2. enter multiplicator try to enter 0,5 doesnt accept the , only accept the . but if I enter 0.5 it doesnt accept it.
3. the value shown when I enter is 1,00 and if I alter the 1 to 0 press arrow right (past the comma) I can enter 5 there to get 0,5

### Actual results

I cant enter a comma in the Multiplicator field, only a dot and that will not work with regional setting to Swedish.
This might happen in other fields as well, but this is the first time I noticed it.

### Expected results

I would like to be able to enter decimal values according to the regional setting, i.e. 0,5 when jumping to the Multiplicator and entering the number from scratch.

### Project file & Debug log uploads

[Dice-test.zip](https://github.com/bambulab/BambuStudio/files/14821566/Dice-test.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/14821567/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MklBambu, 2024-04-30:
Hi, the new version has been released. You can try it out and verify if any issues still exist.
https://github.com/bambulab/BambuStudio/releases/

Bokbacken, 2024-04-30:
Hi,
Unfortunately this made it worse, now I can enter the comma, but it doesn't understand the entered numbers, the only values understood are 0, 1 and 2, no decimal values at all now, doesn't matter if I use , or . as separator; 0.95 or 0,95 is interpreted as 0

MklBambu, 2024-04-30:
I apologize for any inconvenience caused, but in the current version, it is possible to input "0,95" and have it recognized as “0.95” without losing the decimal part.
[image]

Can you provide a video or gif to reproduce your problem?

Thanks!

Bokbacken, 2024-04-30:
[image]
The problem is not what I can enter (anymore) but what gets registered with the Swedish number system, it is dependent on how Windows decode decimal numbers, if I change between , and . as decimal symbol it's clear, as in the animation.
Problem is , it's the Swedish standard (One of our most stupid standards, but it's there anyway)

MklBambu, 2024-05-07:
Thanks! We will fix it in the next version.

Bokbacken, 2024-05-14:
Seems to be working now in the official 1.9.1.66!

Thanks for the correction.

closed 2024-05-14 by Bokbacken