bambulab/BambuStudio #3806: Very High CPU use while idle - Mac OS

icerabbitje, 2024-04-01:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Sonoma 14.4.1

### Additional system information

Apple Studio M1 MAX with 32GB RAM

### Printer

Bambulab x1c 

### How to reproduce

Used my computer yesterday to print a few things. 
It went to sleep. I log on this morning to check my email and see one of my CPUs maxxed out. 
Mac OS said that Bambulab was using 86% CPU ... as I am typing this, it is reporting 101%
What gives?! 

It is not doing anything. Just sitting idle after the last print job yesterday. Printer is asleep ... 
All it has on screen is the UI, prepare tab and the last 3d file on the platform. 

[image]

[image]

### Actual results

Not printing or project related.
Just observing very high CPU use while the program is idle. 

### Expected results

I would expect the program to behave like any other program and calm down when it is not being used. 

### Project file & Debug log uploads

Not printing or project related. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-04-02:
hello @icerabbitje 

thanks for reporting this issue,
is it 100% reproducable on your side?

have you went to the device tab before?

icerabbitje, 2024-04-02:
Well, it was doing it after that long print.
Restarted everything today, with a new print.
It is not acting up currently.

GuzziRaz, 2024-04-02:
See also #2792