bambulab/BambuStudio #3807: Bambu Studio takes a long time to load and the 'Home' page is blank

theboyknowsclass, 2024-04-01:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windws 11

### Additional system information

Device name	WORKSTATION
Processor	Intel(R) Core(TM) i9-10980XE CPU @ 3.00GHz   3.00 GHz
Installed RAM	64.0 GB (63.7 GB usable)
Device ID	63EC69E0-C478-46B3-A5CC-150E8938F20B
Product ID	00342-50652-15031-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

GPU - ASUS 3090 RTX STRIX (GeForce Gameready Driver 551.86)

### Printer

Bambulab X1 C

### How to reproduce

Install BambuStudio
Run

### Actual results

Software takes a long time - showing the 'Loading Configuration..." screen
When it loads (1-2 minutes), the 'Home' Page is blank - so there's no way to login

### Expected results

Software loads quickly
Home page is open allowing login

### Project file & Debug log uploads

[debug_Mon_Apr_01_14_47_29_15412.log.zip](https://github.com/bambulab/BambuStudio/files/14824107/debug_Mon_Apr_01_14_47_29_15412.log.zip)

no project file needed

### Checklist of files to include

- [X] Log file
- [ ] Project file

theboyknowsclass, 2024-04-01:
On subsequent runs, the log files look like this (I've included the network log)

[debug_Mon_Apr_01_14_58_09_9352.log.zip](https://github.com/bambulab/BambuStudio/files/14824145/debug_Mon_Apr_01_14_58_09_9352.log.zip)

theboyknowsclass, 2024-04-01:
seems to be over a minute waiting here 

2024-04-01 14:58:10.861469[Thread 0x00007228]:Slic3r::GUI::Plater::priv::set_current_panel: current_panel 0000000000000000, new_panel 00000191CA32B980
2024-04-01 14:59:32.931887[Thread 0x00007228]:ImageGrid::SetStatus: No printers.

theboyknowsclass, 2024-04-01:
For others with the same problem -  I found a weird workaround that at least let me login - I changed language to dutch - it did a 'Rebuild' which took a long time, then switched it back again... a long wait, and the studio seemed to allow me to login

however when I close the app and reopen, the 'Loading Configuration...' still takes a long time, and the home page no longer works

lanewei120, 2024-04-02:
are you using any special usb keyboard or mouse?
Previously, we have an issue that detecting some keyboard takes a long time

NateHag, 2024-04-08:
I have the same thing happening.
Win11 x64 23H2
AMD 5950X
64GB RAM
Bambu Studio: 1.8.4.51

The app has consistently been very slow to open for me and the home page was always blank UNTIL I unplugged my KVM and started using a different mouse (no keyboard plugged in) for testing what Ianewei120 mentioned.  This could be coincidental, but like I said, it was very consistently slow to open.

lanewei120, 2024-04-09:
for this issue, we also add an option in BambuStudio.Conf to confirm
you can close this app, and edit C:\Users\**user.name**\AppData\Roaming\BambuStudio\BambuStudio.conf manually
add following option
[image]

then save, and relaunch bambu studio

TaylorN15, 2024-04-12:
I am also experiencing this issue. I added the `skip_3dmouse_detect` option to the app settings, but this didn't improve it at all.

lanewei120, 2024-04-13:
> I am also experiencing this issue. I added the `skip_3dmouse_detect` option to the app settings, but this didn't improve it at all.

could you send your log to us for debugging
maybe you have a different issue

TaylorN15, 2024-04-19:
Sure, see [attached](https://github.com/bambulab/BambuStudio/files/15032487/debug_Fri_Apr_19_12_43_25_27844.log.zip)
There seems to be a long delay here:
```
2024-04-19 12:43:34.717371[Thread 0x0000d1e0]:WebView::CreateWebView::<lambda_c148de339aa223f3bcbe68904acff046>::operator (): finished add script message handler for wx.
2024-04-19 12:44:24.794948[Thread 0x0000d1e0]:Slic3r::BackgroundSlicingProcess::select_technology: change the printer technology from  to  
```

RozeDoyanawa, 2024-04-24:
I just purchased a P1S and I have the same issu
e after installing Bambu Studio. Blank home page and long loading time.

Same place-ish for long delay:
[log.zip](https://github.com/bambulab/BambuStudio/files/15100083/log.zip)
`2024-04-24 22:31:01.821979[Thread 0x0000b4f0]:Slic3r::BackgroundSlicingProcess::select_technology: change the printer technology from  to  
2024-04-24 22:31:01.848980[Thread 0x0000b4f0]:Slic3r::GUI::PartPlateList::get_current_slice_result:m_current_plate 0, current_plate 0000022B608404A0
2024-04-24 22:31:01.880980[Thread 0x0000b4f0]:Slic3r::GUI::Plater::priv::set_current_panel: current_panel 0000000000000000, new_panel 0000022B60847C40
2024-04-24 22:32:17.303973[Thread 0x0000b4f0]:ImageGrid::SetStatus: No printers.
2024-04-24 22:32:17.306974[Thread 0x0000b4f0]:monitor: show_status = 4
`

bambu123, 2024-04-25:
How about starting it several times?

TaylorN15, 2024-04-26:
Can you provide me with a secure location to upload the file/provide to you? I don't want to upload it somewhere publicly accessible.

bambu123, 2024-04-26:
> Can you provide me with a secure location to upload the file/provide to you? I don't want to upload it somewhere publicly accessible.

Sorry, we don't have official cloud storage. You can share your cloud storage link by sending a private email

sdo201010, 2024-07-22:
> 对于此问题，我们还在 BambuStudio.conf 中添加了一个选项，以确认您可以关闭此应用程序，并手动编辑 C：\Users\user.name\AppData\Roaming\BambuStudio\BambuStudio.conf 将选项“skip_3dmouse_detect”添加到 true ![图像](https://private-user-images.githubusercontent.com/106916061/320682828-ebf6b6e0-94fe-4aa2-923c-2859082b9bbc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE2MzQ2MzksIm5iZiI6MTcyMTYzNDMzOSwicGF0aCI6Ii8xMDY5MTYwNjEvMzIwNjgyODI4LWViZjZiNmUwLTk0ZmUtNGFhMi05MjNjLTI4NTkwODJiOWJiYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyMlQwNzQ1MzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NGM2MGVlMTY5MGZlNjRkYWU3NzE1OWE1YTAzZGQ0ZTQ2ODg4NWJkMDFhM2YzMjljZGYwMWIyNDNmY2YwZTU5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.LYaXMX71JeX3b3vqndicZLzR3yLnWsONZx76h5MuEV8)
> 
> 然后保存并重新启动 Bambu Studio

按此办法成功解决问题

ericisallred, 2024-08-12:
I had lost the Home page as well as taking forever to load the Bamboo Studio app.  i did get the home page back, buy selecting another language in preferences, for me I switched from France to English, it rebuilt the app / reloaded, then I switched it back to English and the home page was back and functioning.

cardboardpig, 2024-11-02:
I use a KVM and adding `"skip_3dmouse_detect": true` fixed the issue for me.

theboyknowsclass, 2025-01-04:
ok this seems to work now after this config change

harrikauhanen, 2025-02-17:
I also have a KVM (and have a Space Mouse). This trick unfortunately did not work for me. Loading Bambu Studio is unbearably slow. Also, save as... can also take up to a minute to open the save dialog. There are no slow USB-drives the system shall need to refresh when accessing the file system.

rhettprestwich, 2025-05-18:
@lanewei120 Thanks for the tip. I have a KVM (as someone else mentioned) and that worked like a charm. You've saved me so much time.

void-mckenzie, 2026-02-05:
Need to bump this up more. Fix worked for me!

p4n0p71c0n, 2026-08-17:
Still active on the current release. Can you please fix.

wkearney99, 2026-10-20:
For anyone still on the thread, adding "skip_3dmouse_detect": true, with version 2.8.2.61 did nothing to improve the startup time.

Nor did switching from English to another language and back again.  Still takes a while to start Studio.

mrbaggins, 2026-11-06:
Just adding as a recent one, same issues.
Switching from English to French got me a home screen (after a long "reconstruct").
Switching back to English kept it for me, and I logged in. Seems to work fine.
Closing Bambu and reopening it kept "Loading Configurations" for a very long time and it fails to load again, though I've remained logged in.
Adding the skip_3d_mousedetect: true worked. "Loading configurations" took about 3 seconds, and the home screen loaded.