bambulab/BambuStudio #3813: 1.9.0 Beta - Slicing fails with Error Code 127 - command not found

ARSoderberg, 2024-04-03:
### Bambu Studio Version

1.9.0 Beta

### Where is the application from?

Bambu Lab github releases

### OS version

macOS Sonoma 14.4.1

### Additional system information

CPU: Apple M1 Pro, Memory 16GB

### Printer

Bambu Lab P1P upgraded to P1S

### How to reproduce

1. Load attached example .3mf file.
2. Slice plate.

### Actual results

The slicer processes at least 90% then errors out with the following message. Post-processing script Notes - Smooth plate on file /var/folders/ng/ydkffsbn2ql6ndpmrw_yt7lr0000gn/T/bambu_model/Tue_Apr_02/17_59_05#80254#55/Me
tadata/.80254.0.gcode failed.
 Error code: 127 
 Output: /bn/sh: Notes: command not found
 
 
[image]

### Expected results

The file is sliced and ready for printing.

### Project file & Debug log uploads

[error 127 test.3mf.zip](https://github.com/bambulab/BambuStudio/files/14845432/error.127.test.3mf.zip)
[debug_Tue_Apr_02_17_55_57_80202.log.0.zip](https://github.com/bambulab/BambuStudio/files/14845433/debug_Tue_Apr_02_17_55_57_80202.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ARSoderberg, 2024-04-08:
[lanewei120](https://github.com/lanewei120) No, that is not the cause. I have never added any post-processing scripts. I re-checked my submitted test .3mf file and there is no post project script. I have also checked other projects that get this error, and they too do not have any post-processing scripts (no orange circle arrow).

lanewei120, 2024-04-08:
@ARSoderberg 
ok
so what is the printer profile from?
could you show you detailed steps to reproduce it? including create this printer profile

from your 3mf, the post-processing scripts was wrongly set to invalid value

ARSoderberg, 2024-04-17:
[lanewei120](https://github.com/lanewei120) I stand corrected. 

Ok, when comparing process file 'Cooper Medium Production 22@P1P.json' and a few other process profiles compared to those that come built-in, I find the following in the 'post_process' field in the one listed and some other of my .json files.

   "post_process": [
        "",
        "Notes - Smooth plate\n",
        "25 Top-Left Medium\n",
        "25 Standard Medium"
    ],

Note that the text of this field does not show in BambuStudio, that editable field is empty. and in my Beta 3 of BambuStudio 1.9.0.70 the red highlighted circle arrow (from your screensho does not appear for me when using the processs profiles that have the issue.

Upon removing that field and it's data, follow up testing shows that the error 127 does not occur. 

The bug in this case seems to be the empty first line in the post_process field above, it suppresses the display of the text that follows. Whe the "", is removed. The 3 lines of notes do render in the Post-Processing scripts field inside BambuStudio. I have no clue how this content of this field came about. 

Thanks for helping me track down my pain in the neck.

So it seems that this Error 127 'bug' is not one, and this can be closed.

lanewei120, 2024-04-18:
@ARSoderberg thanks for doing this

we will also try to reproduce this on our side
@DanBao-Bambu will have a try

lanewei120, 2024-08-08:
as we can not reproduce it , we can not find why this post-process scripts was changed
close this issue currently and keep on watching
if can reproduce again, we will dig into it

closed 2024-08-08 by lanewei120