bambulab/BambuStudio #3816: Cooling overhang fan speed changes actual layer print speed

UBaer21, 2024-04-03:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Printer

P1S

### How to reproduce

Slicing overhang, with Basic PET 0.08 high quality profile.

### Actual results

[image]

the 10% cooling overhang threshold value causes the overhangs to be printed much faster (or that the overhang speed is used)

### Expected results

[image]

Print speed not be affected for cooling fan speed setting.  The super fast overhang looks terrible!
image from a 25% cooling overhang setting which is now causing that the overhang speed setting is ignored.
The setting should not affect the print speed at all - it should change the fan speed - also it seems the speed settings are ignored
as the outer walls are printed with 8 mm/s and not the 60mm/s from the profile. 

So the root cause is the slow printing speed. Which is then speed up to 60 for the overhang instead of slowed down.

### Project file & Debug log uploads

[BSIssue.zip](https://github.com/bambulab/BambuStudio/files/14857343/BSIssue.zip)
project with BS default profiles
[debug_Wed_Apr_03_20_05_08_14440.log.0.zip](https://github.com/bambulab/BambuStudio/files/14857380/debug_Wed_Apr_03_20_05_08_14440.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2024-04-07:
[image]
this threshold decides if the overhang part could be cooling

UBaer21, 2024-04-07:
Yes so why is the greater overhang printed faster? 
You have here several parameter that interfere, as first the print is slowed down and then the cooling threshold is influencing the print speed again. 
Should that value not just influence the Fan speed?
here a comparison of 10% 25% and 75% (made in version 1.9)
The green stripe shows high speed printing for 10% (default setting Bambu PETG Basic profile)
[image]
with 25% this looks better 
[image]
no change with 75% here
[image]

The "cooling overhang threshold" define when the Overhang speed (print) is used 
[image]

And that setting is faster then the "slow down printing for cooling"  which is causing the speed up.  

A solution would be that the "slow down for cooling" speed setting is used as a max variable for the "overhang speed" so this can not speed up only slow down overhangs.

QingZhangBambu, 2024-04-17:
@UBaer21 Let me briefly explain the process. First, set the speed for the different overhang sections, then slow down to achieve layering time. If the path overhang is less than the cooling overhang threshold, the path will slow down to the layer time. Otherwise, the speed will remain unchanged.

QingZhangBambu, 2024-04-17:
But this option is not easy to understand, and the default value may not be appropriate. I will communicate with relevant personnel to improve it.

UBaer21, 2024-04-17:
Yes but what sense does it make if it is not also slowed down on overhang section - how could it meet the cooling slowed down layer time if the overhang is printed faster?  So if i had a model that has only overhang - the cooling slow down would have no effect.
But i agree if you have set your profile with the correct slow overhang times this would not be an issue. This problem is if you select a different filament like PETg which is slowed down via filament settings but still using the fast default process. The consequence is that you need to change the filament and also create a user preset for PETg (every filament) as process.

QingZhangBambu, 2024-04-17:
yep, it does the problem.

QingZhangBambu, 2024-10-16:
@UBaer21 it has been fixed and will be released in the next version, please wait.

zhangrui1114, 2024-11-13:
[image]
Could u please update the wiki for the new option? How does it work? It's confused.

QingZhangBambu, 2024-11-13:
@zhangrui1114 OK, I've talked to the concerned staff.