bambulab/BambuStudio #3820: Export with vertex color data after painting in Bambu Slicer (any importable format to Blender)

CptJackieSparrow, 2024-04-04:
<?xml version="1.0" encoding="UTF-8"?>
<model unit="millimeter" xml:lang="en-US" xmlns="http://schemas.microsoft.com/3dmanufacturing/core/2015/02" xmlns:slic3rpe="http://schemas.slic3r.org/3mf/2017/06" xmlns:p="http://schemas.microsoft.com/3dmanufacturing/production/2015/06" requiredextensions="p">
 <metadata name="BambuStudio:3mfVersion">1</metadata>
 <resources>
  <object id="1" p:UUID="0000000b-81cb-4c03-9d28-80fed5dfa1dc" type="model">
   <mesh>
    <vertices>
     <vertex x="29.9669991" y="2.79296017" z="-9.76902199"/>
     <vertex x="29.9669991" y="2.78096008" z="-9.61702156"/>
     <vertex x="30.5289993" y="2.83696008" z="-9.76502132"/>
     <vertex x="30.4090004" y="2.82696009" z="-9.76702118"/>
     <vertex x="30.4109993" y="2.82496023" z="-9.72302151"/>
     <vertex x="30.4090004" y="2.82896018" z="-9.77702141"/>
     <vertex x="29.9669991" y="2.69096017" z="-9.00902176"/>
     <vertex x="30.4389992" y="2.72696018" z="-9.0070219"/>
     <vertex x="30.3009987" y="-8.57903957" z="-23.7430019"/>
    </vertices>
    <triangles>
     <triangle v1="0" v2="1" v3="2" paint_color="0C"/>
     <triangle v1="3" v2="4" v3="1"/>
     <triangle v1="3" v2="0" v3="5"/>
     <triangle v1="6" v2="7" v3="8"/>
     <triangle v1="9" v2="10" v3="11"/>
     <triangle v1="12" v2="13" v3="14"/>
     <triangle v1="15" v2="16" v3="12"/>
     <triangle v1="17" v2="18" v3="19" paint_color="0C"/>
     <triangle v1="19" v2="20" v3="17" paint_color="0C"/>
     ...
     <triangle v1="1923" v2="1924" v3="19281"/>
     <triangle v1="19364" v2="2260" v3="2255"/>
     <triangle v1="19364" v2="2255" v3="2256"/>
     <triangle v1="2399" v2="2401" v3="19419" paint_color="0C"/>
     <triangle v1="19420" v2="19421" v3="19419" paint_color="0C"/>
     <triangle v1="19422" v2="19423" v3="19421" paint_color="0C"/>
     <triangle v1="19424" v2="19425" v3="19426" paint_color="0C"/>
     <triangle v1="19423" v2="19426" v3="19425" paint_color="0C"/>
     <triangle v1="19424" v2="19427" v3="19425" paint_color="0C"/>
     <triangle v1="19428" v2="19429" v3="19430" paint_color="0C"/>
     ...
          <triangle v1="30169" v2="27754" v3="27753" paint_color="4C"/>
     <triangle v1="30169" v2="30168" v3="27754" paint_color="4C"/>
     <triangle v1="74559" v2="30090" v3="27775" paint_color="4C"/>
    </triangles>
   </mesh>
  </object>
 </resources>
 <build/>
</model>

So this is a part of xml code that i've extracted from 3mf file i've created in Bambu Studio. I used standart 7zip unarchive and and got the ".model" file (path is: [unarchined .3mf folder in the location]\3D\Objects[filename].stl.model)

I have wast amount of 3d models that i've painted in the Bambu Slicer and I have to render them in Blender realisticly without me having to paint them all over again.

For this, I have to somehow automaticly unarchive 3mf folder, find .model, extract the paint_color information (it might be face/triangle data), convert it to vertex groups or vertex colors, assign materials to those vertex's, and render.

I couldn't find any information online, or any add-on's for this. 3mf importers doesn't work directly unfortunately. They don't preserve vertrex color data.

Grcnalbnz, 2024-08-08:
Have you found any solution ? Thank you

ansonl, 2025-01-03:
Did you find any documentation on the `paint_color` attribute in the 3D data? 

I updated my fork of the Blender 3MF addon to import Bambu 3MF files by recursively importing 3MF components but I don't see any code for `paint_color` that is used in either Bambu or Prusa code.

Brian-David-Oeides, 2025-01-25:
> 1 ... ...
> So this is a part of xml code that i've extracted from 3mf file i've created in Bambu Studio. I used standart 7zip unarchive and and got the ".model" file (path is: [unarchined .3mf folder in the location]\3D\Objects[filename].stl.model)
> 
> I have wast amount of 3d models that i've painted in the Bambu Slicer and I have to render them in Blender realisticly without me having to paint them all over again.
> 
> For this, I have to somehow automaticly unarchive 3mf folder, find .model, extract the paint_color information (it might be face/triangle data), convert it to vertex groups or vertex colors, assign materials to those vertex's, and render.
> 
> I couldn't find any information online, or any add-on's for this. 3mf importers doesn't work directly unfortunately. They don't preserve vertrex color data.

I've tried to attempt the same thing and it seems impossible. Each time I convert the 3MF into a different file format it doesn't seem to save (dae. glb, obj, stl, ply). Have you found any solutions besides having to spend the time painting  it in blender?

CptJackieSparrow, 2025-01-31:
I did not unfortunately