bambulab/BambuStudio #3823: Cannot launch Flatpak

CMDR-Shrine, 2024-04-05:
### Bambu Studio Version

1.9.0 Public beta Flatpak

### Where is the application from?

Bambu Lab github releases

### OS version

Pop!os (linux)

### Additional system information

## System Information
- **Operating System:** Pop!_OS 22.04 LTS x86_64
- **Host:** 82JQ Legion 5 Pro 16ACH6H
- **Kernel:** 6.8.0-76060800daily20240311-generic
- **Uptime:** 19 mins
- **Packages:** 3042 (dpkg), 58 (flatpak)
- **Shell:** bash 5.1.16
- **Resolution:** 2560x1600
- **Desktop Environment (DE):** GNOME 42.5
- **Window Manager (WM):** Mutter
- **WM Theme:** Pop
- **Theme:** Pop-dark [GTK2/3]
- **Icons:** Pop [GTK2/3]
- **Terminal:** gnome-terminal

## Hardware Information
- **CPU:** AMD Ryzen 7 5800H with Radeon Graphics (16) @ 4.463GHz
- **GPU:** NVIDIA GeForce RTX 3070 Mobile / Max-Q
- **Memory:** 3488MiB / 31947MiB

### Printer

Bambu X1 Carbon (not really relivant)

### How to reproduce

1. Open a terminal on your Pop!_OS system.

2. Run the following command to launch BambuStudio via Flatpak:
   
flatpak run com.bambulab.BambuStudio

### Actual results

zac@pop-os:~$ flatpak run com.bambulab.BambuStudio 
zac@pop-os:~$ [2024-04-05 11:23:38.678537] [0x00007d59e0117bc0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1

(process:4): GLib-GObject-CRITICAL **: 11:23:38.702: invalid (NULL) pointer instance

(process:4): GLib-GObject-CRITICAL **: 11:23:38.702: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

### Expected results

The flatpak should open normally. In fact the app image version works perfectly and orca sclicer also funcitons without issues. 

### Project file & Debug log uploads

Not applicable

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-04-07:
we can report flathub issue here
https://github.com/flathub/com.bambulab.BambuStudio/issues

Dagon-, 2024-04-19:
I've run into this as well.
1.9.0 Public Beta 3 

``` /usr/bin/flatpak run --branch=stable --arch=x86_64 --command=entrypoint --file-forwarding com.bambulab.BambuStudio @@u %U @@

 [0x0000765c4dc69bc0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1

(process:4): GLib-GObject-CRITICAL **: 09:15:02.251: invalid (NULL) pointer instance

(process:4): GLib-GObject-CRITICAL **: 09:15:02.251: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

lanewei120, 2024-08-08:
flatpak specific issue, close here

closed 2024-08-08 by lanewei120

matteovidali, 2025-02-24:
This seems to be similar to [https://github.com/bambulab/BambuStudio/issues/2479](https://github.com/bambulab/BambuStudio/issues/2479) in which we have found it to be an issue with the config files - deleting them from `.config` and `.var` folders (where flatpak stores holds them cached) seems to resolve the symptom, though not necessarily solving the underlying issue...