bambulab/BambuStudio #3825: Bambu studio unhandled unknown exception

sparky148, 2024-04-05:
### Bambu Studio Version

v01.08.04.51

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

CPU: intel i9 9900k RAM:16gb 3200mhz ddr4 GPU: Nvidia 3080

### Printer

Bambu x1c 

### How to reproduce

I'm not sure I just opened bambu studio like I usually would

### Actual results

[image]

### Expected results

bambu studio opens as usual

### Project file & Debug log uploads

[debug_Fri_Apr_05_13_13_04_32316.log](https://github.com/bambulab/BambuStudio/files/14885276/debug_Fri_Apr_05_13_13_04_32316.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

sparky148, 2024-04-05:
I think I've found the issue it seems to be an incompatibility with ASUS GPU tweak II

lanewei120, 2024-04-07:
Glad to hear that

so how do you find the root cause, and how to solve it?

sparky148, 2024-04-07:
I just saw in the logs that it was somehow hooking into asus gpu tweak so I closed gpu tweak and bambu studio opened as usual

closed 2024-04-08 by lanewei120