bambulab/BambuStudio #3827: Make the desiccant meter larger and/or include in the help popup

DanTup, 2024-04-06:
This has been looked at before in https://github.com/bambulab/BambuStudio/issues/1120. It was re-raised in https://github.com/bambulab/BambuStudio/issues/2678 but that was closed as a dupe (although I can't find any activate issues, the original one was closed long before).

I still feel like this icon is too small to read, and when you click on it you get a help popup that shows a static image (which is a little misleading, but also a missed opportunity to show a larger version of the graphic IMO).

It seems like there is plenty of space to make this a bit larger and more readable:

[image]

And if not, maybe it could be shown bigger in the help popup?

[image]

RicVilla78, 2024-04-06:
in the beta there is now a drop with a number which even small is easy to read.

[image]

DanTup, 2024-04-06:
Oh that looks much better (although I wonder why the second value to indicate how good it thinks the remaining desiccant is has been removed?).

closed 2024-05-13 by DanTup