bambulab/BambuStudio #3829: Can not rate 5 stars, because Bambu Studio does nog recognise the print as successful.

KRiZ-R, 2024-04-06:
### Bambu Studio Version

1.9.0.60

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

On makerworld, clicked on print profile, then open in Bambu Studio.
Printed the print successfully.
Tried to rate 5 stars, but got the message: At least one successful print record of this print profile is required to give a positive rating (4 or 5 stars)

### Actual results

I could not rate 5 stars.
[image]

### Expected results

I expected to rate 5 stars, because I printed the profile perfectly

### Project file & Debug log uploads

[Shifty V1 - PETG(2).zip](https://github.com/bambulab/BambuStudio/files/14895013/Shifty.V1.-.PETG.2.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

DanBao-Bambu, 2024-04-18:
This situation is due to the unsuccessful printing status obtained when obtaining rating data, which may result in the inability to comment on 4-star or 5-star ratings. We apologize for any inconvenience this may have caused you, but it should be a problem that may not necessarily occur. Have you encountered it again during this period of time? We need your log.

sheehand, 2024-08-31:
I just experienced the same issue on my Bambu Studio on my first print after upgrading to 1.9.5.5.51. Here you can see the same scenario where the print completed successfully, but Bambu Studio refuses to let me post a 5 star review:
[image]

I also confirmed the printer believes the job completed correctly as well:
[image]

I printed a second object (a Bambu Cube) successfully from within the same print profile, which is a trick I have used in the past when I have a failed print that I restart from the server which Bambu Studio doesn't recognize as a successful model print (it should IMHO but I get that may be hard to verify/cross reference). Still the same problem with a block of reporting a 5 star print.

On a whim I closed and re-opened Bambu Studio and went right back into the Device tab w/o loading a print model, and form there I was finally able to successfully report a 5 star print. Not sure what was hung up in Bambu Studio but it required a relaunch which I was nervous about doing because I was worried it wouldn't remember the last print (I haven't acknowledge the successful print on the printer screen yet, so maybe that's why it picked it up?).

DanBao-Bambu, 2024-09-04:
@sheehand I need you to upload a log about when this situation occurred, reference: https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

Goodbill, 2024-10-05:
Same issue

TheRealHotFix, 2024-10-06:
> @sheehand I need you to upload a log about when this situation occurred, reference: https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

I didn't see your response until Goodbill responded. I'm also unclear why I need to upload printer logs for something Bambu Studio is refusing to do. But if it happens again I will upload the logs. At this point I think they probably got pushed out right? I.E. I can't go back to 8/28 in the logs at this point I don't think.

emoritzx, 2025-09-17:
I just ran into this as well. I started a print from Bambu Studio, then canceled it when I saw a first layer issue. Once I cleaned the plate, I restarted the print locally _by selecting the model from the printer's local storage._ Now in the MakerWorld website, it only shows the canceled print, and I cannot rate the print profile.

ajgold, 2025-10-31:
A problem with rating a model 4 or 5 stars can occur in the following situation:

1. Begin printing from Bambu Studio (via the web).
2. Cancel the print from Bambu Studio after a system problem (eg. homing issue, plate not placed, etc.)
3. Re-initialise the print from the printer's console

The successful re-print is then not associated with the model.

nikmrx, 2025-10-31:
When dealing with such a “system problem” (end of filament, in my case) I think it is pretty straightforward and obvious to use the console for the “continue” action: you are next to the printer anyway. 
Why would I have to back to my computer To have a “successful” print?
This is a bug, period.