bambulab/BambuStudio #3835: Handle multi-colour files coming from Solidworks

project-x51, 2024-04-07:
Hi,

I'm not sure how you would do it but it would be amazing if you could create something to allow me to export multi-colour files directly into Bambu Studio from Solidworks.

At the moment I need to repaint all the colours inside Bambu Studio. This is tedious! I know Bambu Studio saves to .3MF files and so does Solidworks. But for some reason the files are not imported into Bambu Studio with colours. 

Perhaps a Solidworks macro, or plugin. Or tell us which file format to export from Solidworks? or work with Solidworks to solve this issue together?

Love you work!

Thanks

Robin

SaltWei, 2024-05-15:
Obj file with color has been supported in 1.9.1 version. 
Step file should be better for solidworks, but not easy work. We are investigating.

Thank you for you suggestion!

abudden, 2025-11-18:
The STEP AP214 format includes colours.  It would be brilliant to be able to import coloured STEP files and have Bambu Slicer import them properly.

The only way I've found to do it at the moment is to:

1. Import the STEP file
2. Click the "Split to Objects" button (which drops everything onto the build plate)
3. Press Ctrl-Z (to lift the objects back up to where they were originally)
4. Manually go through the objects applying colours.

It's rather tedious.