bambulab/BambuStudio #3844: Bambu Studio Quits When Trying to Measure Text Surface To Other Object Surface

djaguiar, 2024-04-08:
### Bambu Studio Version

Beta 1.9.0

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro v22H2 - OS Build 19045.4170

### Additional system information

Intel(R) Core(TM) i9-10900X CPU @ 3.70GHz   3.70 GHz, 32.0 GB (31.7 GB usable), 64-bit operating system, x64-based processor, AMD Radeon Pro WX 2100

### Printer

BambuLab X1C

### How to reproduce

Create text on an object
Ctl-Click object and text
Select measure tool
Click on text top surface

### Actual results

BambuStudio Beta v1.9.0 quits after a few seconds if nothing is done or quits immediately if click on a surface on the object (still in measure mode)

### Expected results

Should provide the distance between the selected text surface and object surface

### Project file & Debug log uploads

[BS_Beta_190_2024_04_08_Measure_Crash_log.zip](https://github.com/bambulab/BambuStudio/files/14906215/BS_Beta_190_2024_04_08_Measure_Crash_log.zip)
[Text_Measure_Move_Test.zip](https://github.com/bambulab/BambuStudio/files/14906220/Text_Measure_Move_Test.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

djaguiar, 2024-04-08:
This appears to be resolved in today's (08 Apr 2024) 1.9.0 BETA 2 release (1.9.0.69); the issue was in BETA 1 (1.9.0.60).

Haidiye00, 2024-04-09:
in BETA 1 (1.9.0.60),I did not reproduce this issue. Can you help record a video? Thank you. Then I will confirm if Beta2 has this issue

djaguiar, 2024-04-09:
Hi, the issuedoes nt occur on BETA 2. Thanks!

closed 2024-05-09 by Haidiye00