bambulab/BambuStudio #3846: 1,9 beta - Changing nozzle size is now an error that doesn't allow user to update nozzle in the Print Plate workflow anymore.

longrackslabs, 2024-04-08:
### Bambu Studio Version

1.9.0.60

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Mint 21.3 Cinnamon

### Additional system information

System:
  Kernel: 5.15.0-100-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: Micro-Star product: MS-7C56 v: 1.0 serial: <superuser required>
  Mobo: Micro-Star model: MPG B550 GAMING PLUS (MS-7C56) v: 1.0 serial: <superuser required>
    UEFI: American Megatrends LLC. v: 1.F0 date: 10/11/2023
Battery:
  Device-1: hidpp_battery_0 model: Logitech MX Keys Wireless Keyboard serial: <filter>
    charge: 50% (should be ignored) rechargeable: yes status: Full
CPU:
  Info: 8-core model: AMD Ryzen 7 5800X bits: 64 type: MT MCP smt: enabled arch: Zen 3 rev: 0
    cache: L1: 512 KiB L2: 4 MiB L3: 32 MiB
  Speed (MHz): avg: 2199 high: 2200 min/max: 2200/3800 boost: enabled cores: 1: 2200 2: 2199
    3: 2198 4: 2200 5: 2199 6: 2200 7: 2199 8: 2200 9: 2198 10: 2199 11: 2200 12: 2199 13: 2199
    14: 2199 15: 2200 16: 2199 bogomips: 121603
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: NVIDIA GM206 [GeForce GTX 950] vendor: PNY driver: N/A pcie: speed: 2.5 GT/s lanes: 16
    bus-ID: 2b:00.0 chip-ID: 10de:1402 class-ID: 0300
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nouveau,vesa
    unloaded: fbdev,modesetting gpu: N/A display-ID: :0 screens: 1
  Screen-1: 0 s-res: 2560x1440 s-dpi: 96 s-size: 677x381mm (26.7x15.0") s-diag: 777mm (30.6")
  Monitor-1: default res: 2560x1440 hz: 93 size: N/A
  OpenGL: renderer: llvmpipe (LLVM 15.0.7 256 bits) v: 4.5 Mesa 23.2.1-1ubuntu3.1~22.04.2
    direct render: Yes
Audio:
  Device-1: NVIDIA GM206 High Definition Audio vendor: PNY driver: snd_hda_intel v: kernel pcie:
    speed: 2.5 GT/s lanes: 16 bus-ID: 2b:00.1 chip-ID: 10de:0fba class-ID: 0403
  Device-2: AMD Starship/Matisse HD Audio vendor: Micro-Star MSI driver: snd_hda_intel v: kernel
    pcie: speed: 16 GT/s lanes: 16 bus-ID: 2d:00.4 chip-ID: 1022:1487 class-ID: 0403
  Sound Server-1: ALSA v: k5.15.0-100-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: Micro-Star MSI
    driver: r8169 v: kernel pcie: speed: 2.5 GT/s lanes: 1 port: f000 bus-ID: 2a:00.0
    chip-ID: 10ec:8168 class-ID: 0200
  IF: enp42s0 state: up speed: 1000 Mbps duplex: full mac: <filter>
Bluetooth:
  Device-1: Broadcom BCM20702A0 Bluetooth 4.0 type: USB driver: btusb v: 0.8 bus-ID: 1-6:5
    chip-ID: 0a5c:21e8 class-ID: fe01 serial: <filter>
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 2.1 lmp-v: 4.0
    sub-v: 220e hci-v: 4.0 rev: 1000
Drives:
  Local Storage: total: 1.95 TiB used: 622.85 GiB (31.2%)
  ID-1: /dev/nvme0n1 vendor: Western Digital model: WDBRPG5000ANC-WRSN size: 465.76 GiB
    speed: 31.6 Gb/s lanes: 4 type: SSD serial: <filter> rev: 111130WD temp: 38.9 C scheme: GPT
  ID-2: /dev/sda vendor: Western Digital model: WD6400AAKS-22A7B2 size: 596.17 GiB
    speed: 3.0 Gb/s type: N/A serial: <filter> rev: 3B01
  ID-3: /dev/sdb vendor: Western Digital model: WD10EZEX-00BN5A0 size: 931.51 GiB
    speed: 6.0 Gb/s type: HDD rpm: 7200 serial: <filter> rev: 1A01 scheme: MBR
Partition:
  ID-1: / size: 365.24 GiB used: 44.67 GiB (12.2%) fs: ext4 dev: /dev/nvme0n1p2
  ID-2: /boot/efi size: 190.5 MiB used: 6.1 MiB (3.2%) fs: vfat dev: /dev/nvme0n1p1
  ID-3: /home size: 915.82 GiB used: 288.67 GiB (31.5%) fs: ext4 dev: /dev/sdb1
Swap:
  ID-1: swap-1 type: partition size: 93.13 GiB used: 846.8 MiB (0.9%) priority: -2
    dev: /dev/nvme0n1p3
USB:
  Hub-1: 1-0:1 info: Hi-speed hub with single TT ports: 10 rev: 2.0 speed: 480 Mb/s
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-2: 1-2:2 info: Genesys Logic Hub ports: 4 rev: 2.0 speed: 480 Mb/s power: 100mA
    chip-ID: 05e3:0608 class-ID: 0900
  Device-1: 1-5:3 info: Logitech Unifying Receiver type: Keyboard,Mouse,HID
    driver: logitech-djreceiver,usbhid interfaces: 3 rev: 2.0 speed: 12 Mb/s power: 98mA
    chip-ID: 046d:c52b class-ID: 0300
  Device-2: 1-6:5 info: Broadcom BCM20702A0 Bluetooth 4.0 type: Bluetooth driver: btusb
    interfaces: 4 rev: 2.0 speed: 12 Mb/s chip-ID: 0a5c:21e8 class-ID: fe01 serial: <filter>
  Device-3: 1-7:7 info: Micro Star MYSTIC LIGHT type: HID driver: hid-generic,usbhid
    interfaces: 1 rev: 1.1 speed: 12 Mb/s power: 500mA chip-ID: 1462:7c56 class-ID: 0300
    serial: <filter>
  Device-4: 1-10:8 info: Kensington Expert Wireless Trackball Mouse (K72359WW) type: Mouse,HID
    driver: hid-generic,usbhid interfaces: 2 rev: 1.1 speed: 12 Mb/s power: 100mA chip-ID: 047d:8018
    class-ID: 0300
  Hub-3: 2-0:1 info: Super-speed hub ports: 4 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003
    class-ID: 0900
  Hub-4: 3-0:1 info: Hi-speed hub with single TT ports: 4 rev: 2.0 speed: 480 Mb/s
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-5: 4-0:1 info: Super-speed hub ports: 4 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003
    class-ID: 0900
Sensors:
  Message: No sensor data found. Is lm-sensors configured?
Repos:
  Packages: 3490 apt: 3454 flatpak: 11 snap: 25
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/1password.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/1password-archive-keyring.gpg] https: //downloads.1password.com/linux/debian/amd64 stable main
  Active apt repos in: /etc/apt/sources.list.d/brave-browser-release.list
    1: deb [signed-by=/usr/share/keyrings/brave-browser-archive-keyring.gpg arch=amd64] https: //brave-browser-apt-release.s3.brave.com/ stable main
  Active apt repos in: /etc/apt/sources.list.d/github-cli.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/githubcli-archive-keyring.gpg] https: //cli.github.com/packages stable main
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/nordvpn.list
    1: deb https: //repo.nordvpn.com/deb/nordvpn/debian stable main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com virginia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
  No active apt repos in: /etc/apt/sources.list.d/playonlinux.list
  No active apt repos in: /etc/apt/sources.list.d/steam-beta.list
  Active apt repos in: /etc/apt/sources.list.d/steam-stable.list
    1: deb [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
    2: deb-src [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
Info:
  Processes: 587 Uptime: 27d 19h 11m wakeups: 1608 Memory: 62.72 GiB used: 22.51 GiB (35.9%)
  Init: systemd v: 249 runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12/9 Client: Cinnamon v: 6.0.4
  inxi: 3.3.13

### Printer

Bambu X1C

### How to reproduce

1. Change nozzle size
2. Slice Plate
3. Print Plate
4. Warning "Your nozzle diameter in sliced file is not consistent with memorized nozzle...
5. New behavior / workflow is innefficeint and uneeded - effectivly treast this as critical error rather than the 1.8 warning with the ability to update the preset/memorized value.

resubmitting as a bug - though the dev team will probably debate this as a new 'feature'.  You can recreate with any project.

### Actual results

1.9 this is now effectively an error, forcing the user to cancel the print attempt, navigate to a new "Printer Parts", set nozzle type and size, then go back to Preview to Print Plate.  This is not a good workflow. Recommend reverting back to 1.8 flow -though it might be better to default to the "Confirm and Update" button rather than the "Confirm"
[image]

### Expected results

In 1.8, this is a warning/notify that allows the user to "Confirm and Update" to sync the nozzle preset. This is an efficent workflow - though it might be better to default to the "Confirm and Update" button rather than the "Confirm"
[image]

### Project file & Debug log uploads

You can recreate with any project.

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2024-04-17:
@longrackslabs 
We think modifying nozzle parameters is a very cautious thing, and users may ignore reminders and perform update nozzle operations. In some cases, this can result in some negative outcomes.

SaltWei, 2024-05-15:
> So you want to penalize advanced users? I change nozzles often - sometimes every print.

After changing nozzles, user can update the information on printer screen. So the printer knows a new nozzle has been installed.

This information is very important for both printer and studio. Since currently there are has no sensor on printer to detect what nozzle has been installed.

On printer size, the nozzle information will be used to filter the pa profiles which is stored in printer, and also be published to studio and app to match the right sliced file.

On app side, this information is also used to filter project which is required to be printed at specific nozzle diameter.

SaltWei, 2024-05-15:
> let me decide if I need it or not and keep the old flow.

We can add one option at studio preference to never receive nozzle information from printer and hide the popup window. 
@walterwongbbl This is not difficult, right?

closed 2025-09-20 by longrackslabs