bambulab/BambuStudio #3847: Can't "Send" a print to printers SDC

GuzziRaz, 2024-04-08:
### Bambu Studio Version

1.9.0.60 and 1.9.0.69

### Where is the application from?

Bambu Lab github releases

### OS version

macOS Sonoma 14.4.1

### Additional system information

- I first saw the problem using Studio 1.9.0.60 and firmware 1.7.2.17.
- Problem still exist after power cycle
- Problem still exist after upgrading to Studio 1.9.0.69 and fw 1.7.3.0.

### Printer

X1C

### How to reproduce

1. Slice any print
2. Try to Send to printer (i.e. send G-code to printer's SDC) from Bambu Studio

### Actual results

Despite the fact I have full lag-free control over the printer, including live video, sending fails immediately every time, with this popup:
[image]

### Expected results

The file should be sent to printer's SDC for later printing.

NOTE that using Orca Slicer (eg. latest release), the G-Code is sent as expected and can be printed from touchscreen. No problem seen at all.

### Project file & Debug log uploads

[send_bug.zip](https://github.com/bambulab/BambuStudio/files/14911284/send_bug.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2024-04-17:
Hi @GuzziRaz 
I found these from the logs you provided. “access code is invalid”. 
Can you refresh the access code of the device again. It is best to delete the BambuStudio.cn file as well.

GuzziRaz, 2024-04-17:
Hi @walterwongbbl. No, that log message is bogus: Like I said, I can print, I can control the printer and filaments or watch video just fine. I just can't "Send" a print to printer's SDC. And it's specific to the Bambu Studio versions I listed. I just now also tested Studio 1.9.0.70 and FW 1.7.3.1 beta - the bug is still present. Orca works fine and **older Bambu Studio versions work fine**.

Here's is a possible clue though: I have noticed that in BambuStudio.conf, these lines:
```
    "access_code": {
        "00M09Axxx100xxx": "deadcafe"
    },
```
They filp between the above (which I reckon is correct), and the below:
```
    "access_code": {
        "00M09Axxx100xxx": "deadcafe\n"
    },
```
That is, the common escape code for a LF is added to the end if the access code. Sometimes when I view the file, it ends with `\n` and sometimes it doesn't. I have seen that for a long time without any problems. But just *maybe* this new problem is somehow related to this.

GuzziRaz, 2024-04-17:
I verified again that if I downgrade to 1.8.4.51, I can "Send" to printer.

That LF diversion in the config doesn't seem to matter: I noticed old Bambu Studio and Orca seem to used the `\n` in the end of the access code string, whereas Bambu Studio beta did not. I tried adding it but it made no difference (Send did NOT work either way, everything else DID work fine either way).

Also BTW I noticed the beta has the access code twice in the json config - once as "access_code" and once as "user_access_code". Only the former lacked the LF.

I also tried logging out from the printer, and logging in again. It didn't help, I can still do everything except Send.

GuzziRaz, 2024-04-24:
This bug is still present with firmware 1.7.3.7 and Studio 1.9.1.56.

Also, I have found that when the "LAN connection failed" window pops up, the access code shown actually does contain a linefeed (not escaped as `\n` but an actual linefeed). If I remove it and retry, Studio hangs until I force quite it.

walterwongbbl, 2024-04-25:
@GuzziRaz 
Very sorry, we have recorded this issue and it will be fixed in the next version.

schaet, 2024-05-28:
Is there any fix for this yet? very frustrating.
I am using Orca Slicer .. tried version 1.6.x, 1.8 and 2.0

bauscha, 2024-06-12:
> Is there any fix for this yet? very frustrating. I am using Orca Slicer .. tried version 1.6.x, 1.8 and 2.0

Same problem for me, even with the new firmware released yesterday

ZeroInputCtrl, 2024-06-16:
I have two printers, they are both up to date. Only one of them has this problem (it's my older one if that). I've noticed that if i remove a character, hit test and let it fail then add the removed character back in, hit test it will connect. Try sending again, during the upload it freezes after 10% BUT if i wait a minute before killing bambu studio i notice the file is on the printer and not corrupted. I've done this with a single plate upload and with a multi-plate upload. So it's working in the backend but something in the application is hanging in the frontend.

ZeroInputCtrl, 2024-06-16:
Quick followup, Just updated to v01.09.02.57. No change in behavior. Did notice if on the LAN Connection Failed Pane, if you click on the access code and arrow to the right it'll look like the code disappears, it's that newline. Deleting the newline has the same effect as what i mentioned in my previous comment. Saves a step but still broken.

GuzziRaz, 2024-06-19:
@walterwongbbl wrote:
> @GuzziRaz Very sorry, we have recorded this issue and it will be fixed in the next version.

Apparently it wasn't fixed, so please ensure it is addressed ASAP.

Skiidlive, 2024-06-22:
What is the status of this issue? Do I need to start a support ticket or is there no fix?  I have 3 X1 machines and one has this issue which is killing my productivity.

brad3d, 2024-07-03:
I can still repro this issue in 1.9.3.50
Was a support ticket ever filed?

magnumripper, 2024-07-04:
They have replied here that [it will be fixed](https://github.com/bambulab/BambuStudio/issues/3847#issuecomment-2076128456). That was months ago as well as a couple of firmware versions ago though.

GuzziRaz, 2024-07-04:
They have replied here that [it will be fixed](https://github.com/bambulab/BambuStudio/issues/3847#issuecomment-2076128456). That was months ago as well as a couple of firmware versions ago though.

sapzxc, 2024-07-08:
Still not working, same issue, can not send

GuzziRaz, 2024-07-10:
I tried deleting my config and recreating one (but then synced config from cloud). The problem remained afterwards. I then did [the workaround mentioned above](https://github.com/bambulab/BambuStudio/issues/3847#issuecomment-2171600399) and initially that seemed to have fixed it, but it hangs indefinitely at "Preparing print job" at 10%.

GuzziRaz, 2024-07-10:
@schaet @bauscha @ZeroInputCtrl @Skiidlive @brad3d @sapzxc are you all on macOS or does anyway see this problem on Windows? I wouldn't touch Windows with a ten foot pole so I can't test it.

Now on macOS Sonoma 14.5 and Studio 1.9.3.50 - problem still there.

sapzxc, 2024-07-11:
> @schaet @bauscha @ZeroInputCtrl @Skiidlive @brad3d @sapzxc are you all on macOS or does anyone see this problem on Windows? I wouldn't touch Windows with a ten foot pole so I can't test it.
> 
> Now on macOS Sonoma 14.5 and Studio 1.9.3.50 - problem still there.

macOS Ventura 13.3.1 (a)

schaet, 2024-07-11:
I am running Windows 10

allanrbo, 2024-08-17:
In my case it was because I forgot to have an SD card inserted.

sapzxc, 2024-08-21:
It looks like the problem is not with the slicer but with the printer firmware. I found that the program uses SFTP over port 990 and tries to upload an archive to the printer. Then I tried to connect to FTP via the command line and found an issue with SSL.

```
lftp -d -c 'open -e "set ftps:initial-prot ""; \
   set ftp:ssl-force true; \
   set ssl:verify-certificate false; \
   set ftp:ssl-protect-data true; \
   ls; " \
   -u "bblp","xxxxxxxx" \
   ftps://192.168.1.xxx:990 '
```

```
<--- 220 (vsFTPd 3.0.3)
---> FEAT
<--- 211-Features:
<---  AUTH TLS
<---  EPRT
<---  EPSV
<---  MDTM
<---  PASV
<---  PBSZ
<---  PROT
<---  REST STREAM
<---  SIZE
<---  TVFS
<---  UTF8
<--- 211 End
---> OPTS UTF8 ON
<--- 200 Always in UTF8 mode.
---> USER bblp
<--- 331 Please specify the password.
---> PASS xxxxxxxx
**** SSL_read: wrong version number
---- Closing connection
```

Also, when I tried to connect via Cyberduck, it timed out.

I tried resetting the firmware, but it didn't help.

One more thing: when I click "close" and then "send" repeatedly, after many clicks, the function eventually works. Sometimes I get an error saying "connection closed".

Maybe this information will help resolve the issue.

ZeroInputCtrl, 2024-08-21:
"It looks like the problem is not with the slicer but with the printer firmware."
There may be two issues present which makes sense as one printer i have works just fine. There is definitely an issue with the software, the entire application shouldn't lock up once the file is sent. See my previous comments regarding the lockup side of the issue.

palopjay, 2024-08-22:
I've same configuration ZeroInputCtrl has (2 printers x1c, same place one above the other) I'm working with 1.9.3.50version. According to the @ZeroInputCtrl workaround, I tried to simply place cursor at the and of the Verification Code (click and press END) and "deleting" a char, it works. File sent and it is printing. Effectively The backspace doesn't delete anything but a possible "phantom space" at the end of verification code that makes it "different" from expectation of the software?

jhogendorn, 2024-09-11:
I've just run into this issue, cannot send to the printer. Its September, the fix was promised in April. Can someone from Bambu update on this?

On the other tickets theres a comment that this feature is now unsupported outside of lan-only mode. Is this the case, and if so, why?

Sidd-86, 2024-09-21:
hey. Late september here, this is still an issue on latest update

Naiki92, 2024-09-28:
I am also still facing the issue. Workaround that worked out for me today:
1. Put printer in "LAN Mode" and generate a new code on the printer
2. Try to reconnect in BambuStudio - it will show it in LAN-Only but fail due to wrong code
3. Type in the new code and connect

After that I could slice and send to the printer. I went to the printer could see the file on MicroSD card and start the print. Then switched off the LAN Mode (But didn't change the code), reconnected again in BambuStudio and sending to the printer works fine again for me now.

Sidd-86, 2024-09-28:
Hey Naiki, thank you for that. Will give it a try, it just might work.

Skiidlive, 2024-09-28:
This worked for me. All of this time dealing with this issue with no fix from bambu. Shame

walterwongbbl, 2024-10-17:
Hi. We have decided to use a new method for file transfer, which should be able to completely solve this problem.

Sidd-86, 2024-10-17:
> Hi. We have decided to use a new method for file transfer, which should be able to completely solve this problem.

That's GREAT news! when is this expected to be implemented?

briodan, 2025-01-19:
> Hi. We have decided to use a new method for file transfer, which should be able to completely solve this problem.

any update on this?

thetechpanda, 2025-01-20:
I've had the same issue, showing "access code is invalid" in the logs. In my case the issue was with duplicated `access_code` keys in `BambuStudio.Conf`, one at the start of the file, one at the very end.

Closed all Bambu Studio windows, removed both entries from the file (removed `access_code` keys and values), restarted Bambu Studio, he asked for the access code once more, but this time worked, I can send files to my P1S.

hope this helps.

Woreec, 2025-01-20:
> > Hi. We have decided to use a new method for file transfer, which should be able to completely solve this problem.
> 
> any update on this?

they are changing data transfer to bambu connect im guessing that was what they meant...

briodan, 2025-01-20:
> > > Hi. We have decided to use a new method for file transfer, which should be able to completely solve this problem.
> > 
> > 
> > any update on this?
> 
> they are changing data transfer to bambu connect im guessing that was what they meant...

All things are possible but this is a function of connecting from Bambu studio to the printer. It would be hilarious and ridiculous at the same time if you need Bambu connect to get a file loaded on the printer from Bambu studio.

brad3d, 2025-04-19:
This is still an issue for me on the new firmware and new software, anyone manage to get it resolved?

palopjay, 2025-04-19:
I resolved in the past releases removing the printer from mi Bambi Studio
and account, removing it from my WiFi network, then re-adding it back. No
more issues. It worked for me, try your side.

Paolo