bambulab/BambuStudio #3848: Crash when attempting to slice

shivachirr, 2024-04-09:
### Bambu Studio Version

1.8.4.54

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

cpu: amd ryzen 9 7950 x3d
gpu: amd radeon rx 7900xtx

### Printer

x1c, a1 mini

### How to reproduce

1. load a project
2. slice any plate
3. then CTD with no error posted

### Actual results

Programs CTD with no errors or reason clear and seems random when it wants to work

### Expected results

the slicer should start processing the object wanted to print

### Project file & Debug log uploads

works with any project and 3d object

### Checklist of files to include

- [ ] Log file
- [ ] Project file

vicar5235, 2024-04-13:
I also encountered the same problem. Simple models won't crash, but slightly more complex software will crash directly, while the Mac version doesn't have the same problem and is waiting for repair

QingZhangBambu, 2024-04-17:
@shivachirr hi, could you upload your 3mf file and dump file?
@vicar5235 is your CPU is Intel i9?

vicar5235, 2024-04-17:
My processor is 14700KF, and my friend's other 14900KF is also experiencing the same problem. When slicing complex models (such as Godzilla), the software will crash directly, and slicing can only be completed normally when slicing simple models (shovels).

vicar5235, 2024-04-17:
> @shivachirr您好，您可以上传您的 3mf 文件和转储文件吗？ @vicar5235你的CPU是Intel i9吗？

My processor is 14700KF, and my friend's other 14900KF is also experiencing the same problem. When slicing complex models (such as Godzilla), the software will crash directly, and slicing can only be completed normally when slicing simple models (shovels).

QingZhangBambu, 2024-04-17:
@vicar5235 could you upload your 3mf files and dump files? I will check this issue.

vicar5235, 2024-04-17:
> @vicar5235 could you upload your 3mf files and dump files? I will check this issue.

https://makerworld.com/zh/models/189250

QingZhangBambu, 2024-04-17:
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue @vicar5235 @shivachirr you can get dump and log files as step of the guide

vicar5235, 2024-04-17:
> https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue @vicar5235 @shivachirr you can get dump and log files as step of the guide

[Uploading bambu-studio.exe.28236.dmp…]()

vicar5235, 2024-04-17:
[bambu-studio.exe.28236.dmp](https://github.com/bambulab/BambuStudio/files/15008383/bambu-studio.exe.28236.dmp)

vicar5235, 2024-04-17:
> https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue @vicar5235 @shivachirr you can get dump and log files as step of the guide

Uploaded

QingZhangBambu, 2024-04-18:
[image]
@vicar5235 it seems that it crash on clipper lib. a bit tricky to solve it.
if you rotate the model a bit, will the model be sliced sucessfully?

vicar5235, 2024-04-18:
> ![image](https://private-user-images.githubusercontent.com/112676417/323550800-0068e957-a0b6-4b3a-8c59-ebce68ce415d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

I just tried rotating the model, but it was not successful. The software still crashes halfway through the slicing process.

QingZhangBambu, 2024-04-18:
@vicar5235 could you try these methods to see if they work for you? some people confirmed useful https://github.com/bambulab/BambuStudio/issues/3318

chrille77, 2024-06-12:
I have the same issues. I have an Intel i9-14900KF, and when slicing complex models Bambu Studio will crash. A workaround is to reduce the amount of CPU cores that Bambu Studio can use, I change the CPU affinity to only include the first 8 cores.
[image]