bambulab/BambuStudio #3849: Latest update BambuStudio for Linux missing setup for Bambu Printers

DennisJKelley, 2024-04-09:
### Bambu Studio Version

1.9.0.69

### Where is the application from?

Bambu Lab github releases

### OS version

LMDE 6 Faye 6.04 Kernel 6.1.0-18-amd64

### Printer

Bambu Lab P1S Combo

### How to reproduce

Setup Printer ... NO BAMBU PRINTERS are listed.

Go to Setup Printer ... enter correct IP .... error message: could not connect to OctoPrint: curl:Couldn't connect to server:
Failed to connect to 10.0.0.xxx port 80 after 129 ms: Couldn't connect to server
[Error 7]
Note: OctoPrint version at least 1.1.0 is required.

### Actual results

NO BAMBU PRINTERS are listed in Printer Setup

### Expected results

Bambu Printers should appear in the printer selection windows.  My P1S and Bambu Studio was running fine until this latest update was made this afternoon.

### Project file & Debug log uploads

/home/master/.var/app/com.bambulab.BambuStudio/config/BambuStudioInternal/log/log.tar.gz

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2024-04-10:
@DennisJKelley 
hello
are you using the ubuntu image or fedora image?

DennisJKelley, 2024-04-10:
Ubuntu

DennisJKelley, 2024-04-10:
> can not find your logs, something wrong when you uploading?
>
> we have just tried it on ubuntu 22
> and can work well here
> image.png (view on web)<https://github.com/bambulab/BambuStudio/assets/106916061/d4db7968-be1f-42c0-a806-f4ecc2b10a1d>
>
> could you remove the directory of ~/.config/BambuStudioBeta and try again

I completely Uninstalled, re download and reinstalled.  Seems OK now.. Linux Ubuntu Beta Version 1.9.0.69.  Plugin version 01.09.00.03

lanewei120, 2024-04-11:
thanks for verifying

closed 2024-04-11 by lanewei120