bambulab/BambuStudio #3850: network

maxalford, 2024-04-09:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11 pro

### Additional system information

asus Z690-g
120 gb memory
12th gen i9 1290k
amd radion rx6900xt
intel LGA 1700 for 13th gen

### Printer

bambu lab x1c

### How to reproduce

start printer after 5 minutes the network turns itself off and has to be restarted manually
[image]

### Actual results

it continually drops the network and i can't upload a file to print i have reset the printer and my home network and it is still doing it i also have an a1 mini that is not having any problems with the conections

### Expected results

network card is faulty

### Project file & Debug log uploads

[bambu-studio.exe.5496.zip](https://github.com/bambulab/BambuStudio/files/14913131/bambu-studio.exe.5496.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-04-10:
@maxalford 
it seems a printer related issue
could you upload the printer related logs?

maxalford, 2024-04-10:
[bambu-studio.exe.5496.zip](https://github.com/bambulab/BambuStudio/files/14926774/bambu-studio.exe.5496.zip)
the log files

ChloeShang123, 2024-04-10:
Hello maxalford,
Sorry to hear drops of network.
Could you export device logs through sd card and upload it through website? And leave us your ticket number here.
Some reason may lead wifi issues:
1. the wifi module parts loosed.
2. firmware or other software issues(maybe routers? netword environment? firewall settings). Could you show us your firmware version??This may be analyzed by devices logs.
3. other unknown issues.
Our custom service team can help you if hardware has some problems or software if there is somehow way to resume. 
Thanks very much.

maxalford, 2024-04-10:
i reset the router and in fact all the network i also have an A1 mini sitting beside the x1c and have no network problems with it i updated the firmware yesterday to the most recent and this is when i thought the problem occured so i downgraded the firmware useing the handy app version 01.07.02.00

ChloeShang123, 2024-04-10:
> i reset the router and in fact all the network i also have an A1 mini sitting beside the x1c and have no network problems with it i updated the firmware yesterday to the most recent and this is when i thought the problem occured so i downgraded the firmware useing the handy app version 01.07.02.00

Maybe...forget the wifi, then connect it again??
Could you try this way??

maxalford, 2024-04-10:
i did a reset of the printer and new conection still the same no network at all now

ChloeShang123, 2024-04-10:
> i did a reset of the printer and new conection still the same no network at all now

So sad....
Now only way to analyze is to upload device logs firstly... Let R&D team to anaylze reason in details...Maybe they will give a solution to solve your problem.

maxalford, 2024-04-10:
i have uploaded the device logs

ChloeShang123, 2024-04-10:
> i have uploaded the device logs

Could you give me your ticket number or load link??

maxalford, 2024-04-10:
the log i uploaded was the studio log sorry just getting device log now

maxalford, 2024-04-10:
the log is to big to upload

maxalford, 2024-04-10:
https://drive.google.com/file/d/1bzw292HvuT1dHD2WDCazz84L4Kc8Ik_u/view?usp=sharing

ChloeShang123, 2024-04-10:
got it thanks!!

maxalford, 2024-04-12:
any progress with this yet

shuizhiguo, 2024-04-15:
we add you to private test group in this test stage, reboot your machine, you will see new version 1.7.3.1 , this version optimizes wifi feature, if it doesn't solve this issue, please change AP main board from contact after-sales